h2zero/NimBLE-Arduino #501: peer_addr = null when doing directed advertising

ViktorRindom, 2023-01-31:
When I do "BLE_GAP_CONN_MODE_UND" everything works but when I change to "BLE_GAP_CONN_MODE_DIR" I always get the error "Unable to advertise - Duration too long". I have found out that this is a result of the rc value of 3 (BLE_HS_EINVAL). This error code is used in a lot of places and I started replacing the values with my own unique error codes and found out that my error was peer_addr being null in the file ble_gap.c in the function ble_gap_adv_validate. But i cant fix this since the file NimBLEAdvertising.cpp puts null in the call ble_gap_adv_start. 

TLDR: NimBLEAdvertising.cpp sets peer_addr to null and ble_gap.c returns an error if peer_addr is null

ViktorRindom, 2023-01-31:
the line with null: 
https://github.com/h2zero/NimBLE-Arduino/blob/0d9f039ba7f960946a8c9aeb639c3a64294f0d60/src/NimBLEAdvertising.cpp#L627
the line with null check:
https://github.com/h2zero/NimBLE-Arduino/blob/0d9f039ba7f960946a8c9aeb639c3a64294f0d60/src/nimble/nimble/host/src/ble_gap.c#L2614

h2zero, 2023-02-04:
Thanks for bringing this up, a new parameter should be added to advertising start in this case that takes the address.

ViktorRindom, 2023-02-04:
Is it possible to do directed advertise to a list and not a single peer_addr?

h2zero, 2023-02-04:
Only a single address can be directly advertised to.

ViktorRindom, 2023-02-04:
It would also be nice if all errors did not use rc code 3

ViktorRindom, 2023-02-06:
@h2zero will you write here when the bug is fixed?
So we can close this issue and so I can update to the fix?

h2zero, 2023-02-20:
@ViktorRindom please check #507 and let me know if this works for you.

ViktorRindom, 2023-02-22:
Im using Platformio how do I use the pull request version of the library?

h2zero, 2023-02-22:
You can just use the branch name: `lib_deps = https://github.com/h2zero/NimBLE-Arduino.git#bugfix-directed-adv`

PR #507 (2023-04-06): Add directed peer address parameter to advertising start.
Adds a parameter to NimBLEAdvertising::start to take the peer address for use with directed advertising.

Fixes #501
closed 2023-04-06 by h2zero