h2zero/NimBLE-Arduino #502: Service Change Indication

Koxx3, 2023-02-03:
Hello,

I have troubles with Android phones when adding/removing characteristics or services with bonded devices.
It seems a common issue, and I found a lot of users with the same troubles.
I tried everything on the "client side" (Android cache refresh hack ... since it's not a public API), but it still doesn't work.

I found that BLE seems to already include "a flag" to indicate the client it needs to refresh his cache... the "Service Change Indication".

It doesn't seems NimBLEServer::serviceChanged() does the same thing (and it's not in the public API).
https://github.com/h2zero/NimBLE-Arduino/blob/release/1.4/src/NimBLEServer.cpp#L168

Espressif seems to offer the API :
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/bluetooth/esp_gatts.html#_CPPv444esp_ble_gatts_send_service_change_indication13esp_gatt_if_t13esp_bd_addr_t

Could it be possible to offer an API for this in NimBLE please ?

More info here :
https://stackoverflow.com/questions/62155016/bluez-and-service-characteristics-cache-issue-with-android

Koxx3, 2023-02-03:
it seems related to the 0x2A05 characteristic

Koxx3, 2023-02-04:
To reproduce :
- ESP32 app with 20 characterics
- launch a android client and bond the device
- update the ESP32 app and insert a new characteristic before all others
- launch a android client => the application is lost and can't read the characterics correctly

h2zero, 2023-02-04:
For this you could try using the lower level call `ble_svc_gatt_changed(uint16_t start_handle, uint16_t end_handle)` with the handle params being 0x0000 and 0xFFFF.

Koxx3, 2023-02-04:
thanks for your answer. 
I finally managed to get it working by forcing android to prune his cache (some hidden API ... ).

closed 2023-02-04 by Koxx3