h2zero/NimBLE-Arduino #505: How to implement a persistent bonding to maintain device bonds across ESP32 reboots and reduce scan and connection time

agualdron, 2023-02-14:
I'm using an ESP31-WROOM-32...
the provided examples are working fine, but I need to config a client that connects to a server (with known UUID) immediately after deep-sleep wake-up to change (ASAP) the value of a server characteristic then go to deep-sleep again. I believe this task can be done in about 1 to 2 seconds. 

Using the provided examples as templates, with a scan time of 1sec, the complete task takes 8 seconds on average in complete.

I'm thinking that If the client information and advertising device information are stored on NVM, the time from reset to change the characteristic value could be reduced considerably.

How to implement a persistent bonding to maintain device bonds across ESP32 reboots and reduce scan and connection time?

doudar, 2023-02-17:
I'm playing around with faster scanning and connections right now, and it looks like you can get the scan time down significantly if you split this up into separate tasks. Eg:
Task one starts a 1 second scan. 
Task two stops the scan some ms later with NimBLEDevice::getScan()->stop();
It's far from perfect (I'm working on reconnecting HID clients which automatically time out), but I am getting reconnects down to just a couple seconds this way.

h2zero, 2023-02-20:
> I'm thinking that If the client information and advertising device information are stored on NVM, the time from reset to change the characteristic value could be reduced considerably.

> How to implement a persistent bonding to maintain device bonds across ESP32 reboots and reduce scan and connection time?

You could try to store the client data and read it on reset, that will reduce the communication time on connection.
Also you can use faster connection parameters, E.g `pClient->setConnectionParams(6, 6, 0, 200);`

agualdron, 2023-02-25:
@doudar  and @h2zero Thank you for your response.

I combined your suggestions, the connection time droped from 8 to 1 second, which is just what the application required. There was no need to store client data to memory, as the scan/connection is now faster enough to do all that in less than a second.

Thank you for your help.

closed 2023-03-16 by agualdron