h2zero/NimBLE-Arduino #510: presence detection

mkohns, 2023-02-26:
Have a question. One important feature for smart home applications is to detect presence of persons (or their phones).
Due to changing random MAC addresses of i.e smart phones, this is not possible with ble scan of advertises.
I mean - this is also a good privacy protection of getting tracked - so don´t get me wrong.

Currently - the only good way which is known to me is using hcitool.
With i.e 
```
hcitool name MAC 
´´´

You can try to get the name of the static "real" MAC address. You can get also more info like vendor, rssi ...
All you need is to have the real mac, which you can get from the phones "about"  menu.

So for presence, this known macs are available - and this is nice.
I think these calls are called "name request" - but I am not sure.

Is something like this possible with NimBLE?

mkohns, 2023-02-27:
Another idea would be to offer a service from within the presence monitor peripheral. Once paired and trusted, the IRK could be used to decode the advertised random MAC to real mac. What about this idea?

https://www.bluetooth.com/blog/bluetooth-technology-protecting-your-privacy/

mkohns, 2023-02-27:
This is what hcidump tells you if you make a name request to a known (NOT Advertised) real MAC

```
hcitool name MAC

< HCI Command: Remote Name Request (0x01|0x0019) plen 10
    bdaddr DC:52:85:F2:82:30 mode 2 clkoffset 0x0000
> HCI Event: Command Status (0x0f) plen 4
    Remote Name Request (0x01|0x0019) status 0x00 ncmd 1
> HCI Event: Remote Host Supported Features Notification (0x3d) plen 14
    bdaddr DC:52:85:F2:82:30
    Features: 0x0f 0x00 0x00 0x00 0x00 0x00 0x00 0x00
> HCI Event: Remote Name Req Complete (0x07) plen 255
    status 0x00 bdaddr DC:52:85:F2:82:30 name 'HERE IS THE NAME'
```

h2zero, 2024-06-04:
NimBLE cannot get the static mac address unless bonded with the phone to get the IRK.

closed 2024-06-04 by h2zero