h2zero/NimBLE-Arduino #511: preffered phy

chegewara, 2023-02-28:
Hi,
i know its not a problem with this library, just with nimble itself, but for others i am reporting it here.
This code will report always 1M_PHY, even if connection is with 2M_PHY:
```
uint8_t rx_phy, tx_phy;
ble_gap_read_le_phy(desc->conn_handle, &tx_phy, &rx_phy);
Serial.printf("RX: %d, TX: %d\n", rx_phy, tx_phy);
```

h2zero, 2023-02-28:
Thanks for pointing this out. I looked at the code and it requests the info from the controller, may be a bug in the espressif controller and not NimBLE. Do you know if this works properly with Bluedroid?

chegewara, 2023-02-28:
No idea, i didnt test it with bluedroid.

btw, i tested ota over ble with 2M_PHY and data are sent way too fast, the code has no chance to write on flash in time

h2zero, 2023-03-01:
> btw, i tested ota over ble with 2M_PHY and data are sent way too fast, the code has no chance to write on flash in time

Cool, I'll have to try that.