h2zero/NimBLE-Arduino #512: NimBLERemoteCharacteristic::writeValue blocks infinitely

kinimodkoch, 2023-02-28:
Hello,
I have a problem with the function "NimBLERemoteCharacteristic::writeValue" which I call multiple times quickly after each other. When my data length does not exceed the MTU size in each write, this works fine. If it exceeds MTU size, a "long wirte" is performed and the following happens:
(1) The first few calls of "writeValue" return "false", but the data is written to the server correctly - I tracked this down to the error code "BLE_ATT_ERR_UNLIKELY 0x0e"
(2) The next call of "writeValue" still returns "false", but now the error code is "BLE_ATT_ERR_PREPARE_QUEUE_FULL 0x09"
(3) The next call of "writeValue" blocks infinitely

Details about (3):
NimBLE seems to trigger the write in NimBLERemoteCharacteristic.cpp line 794:
            rc = ble_gattc_write_flat(pClient->getConnId(), m_handle,
                                      data, length,
                                      NimBLERemoteCharacteristic::onWriteCB,
                                      &taskData);
then it will wait for a notification in line 814:
        ulTaskNotifyTake(pdTRUE, portMAX_DELAY);
This notification is supposed to be done through the callback NimBLERemoteCharacteristic::onWriteCB but never comes.

I use an Arduino environment on an ESP32 with NimBLE 1.4.

I have the following questions:
- Is it a bug that the function blocks infinitely, or misuse of the function?
- How can I distinguish (1) and (2), since "writeValue" just returns "false" in any case and does not pass through the error code?
- Is there a buffer size I can increase to resolve the problem happening in case (2)?

Thanks for your help!

h2zero, 2023-02-28:
Thank you for reporting this issue.

> Is it a bug that the function blocks infinitely, or misuse of the function?

That should not happen, something isn't triggering an event somewhere as the callback should always be invoked. I would need to reproduce this to find the cause.

> Is there a buffer size I can increase to resolve the problem happening in case (2)?

Try changing this to 20 or more:
https://github.com/h2zero/NimBLE-Arduino/blob/0d9f039ba7f960946a8c9aeb639c3a64294f0d60/src/nimconfig.h#L131

kinimodkoch, 2023-03-01:
Thanks for your quick reply!
Unfortunately, increasing CONFIG_BT_NIMBLE_MSYS1_BLOCK_COUNT did not solve my issue.
I noticed that with another server, I can not reproduce this issue. I will investigate further if there is something wrong on the server side.

kinimodkoch, 2023-03-02:
I discovered the root cause, the receiving buffer on the server side was too full. It looks as if the writeValue() function has no way to recover in this case, it keeps blocking, even if the receive buffer is free again, even after the connection is terminated (the callback that the connection is closed is still called).
I have the option to solve my issue on the server side, so solving this bug is not urgent from my side.