h2zero/NimBLE-Arduino #514: rc = 530 on disconnect

ViktorRindom, 2023-03-08:
I want to make a device that only accepts devices previously added to the white list. when a device that is not on the white list connects it should immediately disconnect and continue advertising. I have gotten the whitelist check to work but when I try to disconnect non whitelisted devices I always get "E NimBLEServer: ble_gap_terminate failed: rc=530 Invalid HCI Command Parameters". I have tried the following disconnects in "void onConnect(NimBLEServer *pServer, ble_gap_conn_desc *desc)":
pServer->disconnect(desc->conn_handle, BLE_HS_EREJECT);
pServer->disconnect(desc->conn_handle, BLE_HS_EUNKNOWN);
pServer->disconnect(desc->conn_handle, 0);

TLDR: All I need to know is what parameters I need to disconnect the connected device no immediately.

h2zero, 2023-03-08:
I suggest not setting the second parameter, the issue you're seeing is that the reason code you're giving is not valid.

ViktorRindom, 2023-03-13:
I could not see that you could run it with one parameter. But yeah that compiled im currently testing it. ill close the issue when it passes the tests.

closed 2023-03-22 by ViktorRindom