h2zero/NimBLE-Arduino #518: Value of CONFIG_BT_NIMBLE_NVS_PERSIST is ignored by compiler

Staars, 2023-03-25:
In esp_nimble_cfg.h the existence of a #define for `CONFIG_BT_NIMBLE_NVS_PERSIST` is checked, but the value is ignored, so (at least for my compiler) this is sets `MYNEWT_VAL_BLE_STORE_CONFIG_PERSIST` to 1.
I am not totally sure, if this is undefined behavior and my work for some compilers or if it is completely wrong.

For me this change works:
```
#ifndef MYNEWT_VAL_BLE_STORE_CONFIG_PERSIST
#if     CONFIG_BT_NIMBLE_NVS_PERSIST==1
#define MYNEWT_VAL_BLE_STORE_CONFIG_PERSIST (1)
#else
#define MYNEWT_VAL_BLE_STORE_CONFIG_PERSIST (0)
#endif
#endif
```

h2zero, 2023-03-25:
Thanks, I'll have a look into this.

h2zero, 2023-03-26:
You're correct, I will update this shortly.

PR #520 (2023-03-26): Fix CONFIG_BT_NIMBLE_NVS_PERSIST value not being used.
Fixes #518
closed 2023-03-26 by h2zero