h2zero/NimBLE-Arduino #521: Failed to get remote characteristic

sergem155, 2023-03-29:
Hi!

I am trying to talk to a plant monitor over BLE using ESP32.
I am able to talk to it from [Raspberry PI, using Bluepy](https://github.com/sergem155/miflora-influxdb/blob/master/poll-insert.py) (using handles, though).
With NimBLE, while some characteristics are working fine, attempts to read others fail.

The code that connects, gets a value from one characteristic, then fails to get another is below, along with gatttool output:

```
#include "NimBLEDevice.h"

void setup() {
  Serial.begin(115200);
  Serial.println("Hello World!");
  BLEDevice::init("");
  BLEDevice::setPower(ESP_PWR_LVL_P7);

  BLEClient* floraClient = BLEDevice::createClient();
  BLERemoteService* floraService = nullptr;
  static BLEUUID serviceUUID("00001204-0000-1000-8000-00805f9b34fb");
  static BLEUUID uuid_version_battery("00001a02-0000-1000-8000-00805f9b34fb");
  static BLEUUID uuid_sensor_data("00001a01-0000-1000-8000-00805f9b34fb");
  static BLEUUID uuid_write_mode("00001a00-0000-1000-8000-00805f9b34fb");
  static BLEUUID uuid_timer("00001a12-0000-1000-8000-00805f9b34fb");
  static BLEUUID uuid_history_control("00001a10-0000-1000-8000-00805f9b34fb");
  static BLEUUID uuid_history_read("00001a11-0000-1000-8000-00805f9b34fb");
  BLERemoteCharacteristic* floraCharacteristic;

  //  BLEAddress floraAddress("c4:7c:8d:66:34:90"); // rose
  BLEAddress floraAddress("c4:7c:8d:63:c7:ea"); // bamble Espressi 0c:b8:15 - 77:1a:2a
  if (!floraClient->connect(floraAddress)) {
    Serial.println("- Connection failed, skipping");
    return;
  }

  if(!floraClient->discoverAttributes()){
    Serial.println("- attributes failed to get discovered");
    return;
  }else{
    Serial.println("- attributes discovered");
  }

  floraService = floraClient->getService(serviceUUID);
  if (floraService == nullptr) {
    Serial.println("- Failed to find data service");
    return;
  }else {
    Serial.println("- Found data service");
  }

  floraCharacteristic = nullptr;
  std::string value;
  // get the device battery characteristic -------------
  Serial.println("- Access battery characteristic from device");
  floraCharacteristic = floraService->getCharacteristic(uuid_version_battery);
  if (floraCharacteristic == nullptr) {
    Serial.println("-- Failed, skipping battery level");
    return;
  }

  // read characteristic value
  Serial.println("- Read value from characteristic");
  value = floraCharacteristic->readValue();
  const char *val2 = value.c_str();
  int battery = val2[0];
  Serial.print("-- Battery: ");
  Serial.println(battery);

  Serial.println("- get history control");
  floraCharacteristic = nullptr;
  floraCharacteristic = floraService->getCharacteristic(uuid_history_control);
  if (floraCharacteristic == nullptr) {
    Serial.println("-- Failed, skipping device");
    return;
  }else{
    Serial.println("-- Got the characteristic");
  }
}

void loop() {
}
```

output:
```
18:46:40.861 -> Hello World!
18:46:46.266 -> - attributes discovered
18:46:46.266 -> - Found data service
18:46:46.266 -> - Access battery characteristic from device
18:46:46.266 -> - Read value from characteristic
18:46:46.361 -> -- Battery: 74
18:46:46.361 -> - get history control
18:46:47.151 -> -- Failed, skipping device
```

gatttool:
```
pi@sub:~ $ gatttool --device=c4:7c:8d:63:c7:ea --characteristics
handle = 0x0002, char properties = 0x02, char value handle = 0x0003, uuid = 00002a00-0000-1000-8000-00805f9b34fb
handle = 0x0004, char properties = 0x02, char value handle = 0x0005, uuid = 00002a01-0000-1000-8000-00805f9b34fb
handle = 0x0006, char properties = 0x0a, char value handle = 0x0007, uuid = 00002a02-0000-1000-8000-00805f9b34fb
handle = 0x0008, char properties = 0x02, char value handle = 0x0009, uuid = 00002a04-0000-1000-8000-00805f9b34fb
handle = 0x000d, char properties = 0x22, char value handle = 0x000e, uuid = 00002a05-0000-1000-8000-00805f9b34fb
handle = 0x0011, char properties = 0x1a, char value handle = 0x0012, uuid = 00000001-0000-1000-8000-00805f9b34fb
handle = 0x0014, char properties = 0x02, char value handle = 0x0015, uuid = 00000002-0000-1000-8000-00805f9b34fb
handle = 0x0016, char properties = 0x12, char value handle = 0x0017, uuid = 00000004-0000-1000-8000-00805f9b34fb
handle = 0x0018, char properties = 0x08, char value handle = 0x0019, uuid = 00000007-0000-1000-8000-00805f9b34fb
handle = 0x001a, char properties = 0x08, char value handle = 0x001b, uuid = 00000010-0000-1000-8000-00805f9b34fb
handle = 0x001c, char properties = 0x0a, char value handle = 0x001d, uuid = 00000013-0000-1000-8000-00805f9b34fb
handle = 0x001e, char properties = 0x02, char value handle = 0x001f, uuid = 00000014-0000-1000-8000-00805f9b34fb
handle = 0x0020, char properties = 0x10, char value handle = 0x0021, uuid = 00001001-0000-1000-8000-00805f9b34fb
handle = 0x0024, char properties = 0x0a, char value handle = 0x0025, uuid = 8082caa8-41a6-4021-91c6-56f9b954cc34
handle = 0x0026, char properties = 0x0a, char value handle = 0x0027, uuid = 724249f0-5ec3-4b5f-8804-42345af08651
handle = 0x0028, char properties = 0x02, char value handle = 0x0029, uuid = 6c53db25-47a1-45fe-a022-7c92fb334fd4
handle = 0x002a, char properties = 0x0a, char value handle = 0x002b, uuid = 9d84b9a3-000c-49d8-9183-855b673fda31
handle = 0x002c, char properties = 0x0e, char value handle = 0x002d, uuid = 457871e8-d516-4ca1-9116-57d0b17b9cb2
handle = 0x002e, char properties = 0x12, char value handle = 0x002f, uuid = 5f78df94-798c-46f5-990a-b3eb6a065c88
handle = 0x0032, char properties = 0x0a, char value handle = 0x0033, uuid = 00001a00-0000-1000-8000-00805f9b34fb
handle = 0x0034, char properties = 0x1a, char value handle = 0x0035, uuid = 00001a01-0000-1000-8000-00805f9b34fb
handle = 0x0037, char properties = 0x02, char value handle = 0x0038, uuid = 00001a02-0000-1000-8000-00805f9b34fb
handle = 0x003b, char properties = 0x02, char value handle = 0x003c, uuid = 00001a11-0000-1000-8000-00805f9b34fb
handle = 0x003d, char properties = 0x1a, char value handle = 0x003e, uuid = 00001a10-0000-1000-8000-00805f9b34fb
handle = 0x0040, char properties = 0x02, char value handle = 0x0041, uuid = 00001a12-0000-1000-8000-00805f9b34fb
```

thank you in advance!
Sergey

h2zero, 2023-03-29:
Could you enable logging with the level set to debug and post the log of this please?

sergem155, 2023-03-30:
Here it is - thanks
```
17:50:00.665 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:50:00.698 -> configsip: 0, SPIWP:0xee
17:50:00.698 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:50:00.698 -> mode:DIO, clock div:1
17:50:00.698 -> load:0x3fff0030,len:1344
17:50:00.698 -> load:0x40078000,len:13924
17:50:00.698 -> ho 0 tail 12 room 4
17:50:00.698 -> load:0x40080400,len:3600
17:50:00.698 -> entry 0x400805f0
17:50:00.928 -> [    28][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
17:50:00.960 -> Hello World!
17:50:01.192 -> I NimBLEDevice: BLE Host Task Started
17:50:01.223 -> I NimBLEDevice: NimBle host synced.
17:50:01.223 -> D NimBLEDevice: >> setPower: 7 (type: 11)
17:50:01.223 -> D NimBLEDevice: << setPower
17:50:01.223 -> D NimBLEClient: >> connect(c4:7c:8d:63:c7:ea)
17:50:02.778 -> D NimBLEClient: Got Client event 
17:50:02.778 -> I NimBLEClient: Connected event
17:50:02.909 -> D NimBLEClient: Got Client event 
17:50:02.909 -> I NimBLEClient: mtu update event; conn_handle=0 mtu=23
17:50:02.909 -> I NimBLEClient: Connection established
17:50:02.909 -> D NimBLEClient: >> deleteServices
17:50:02.909 -> D NimBLEClient: << deleteServices
17:50:02.909 -> D NimBLEClientCallbacks: onConnect: default
17:50:02.942 -> D NimBLEClient: << connect()
17:50:02.942 -> D NimBLEClient: >> deleteServices
17:50:02.942 -> D NimBLEClient: << deleteServices
17:50:02.942 -> D NimBLEClient: >> retrieveServices
17:50:03.040 -> D NimBLEClient: Service Discovered >> status: 0 handle: 1
17:50:03.072 -> D NimBLERemoteService: >> NimBLERemoteService()
17:50:03.072 -> D NimBLERemoteService: << NimBLERemoteService(): 0x1800
17:50:03.072 -> D NimBLEClient: Service Discovered >> status: 0 handle: 12
17:50:03.072 -> D NimBLERemoteService: >> NimBLERemoteService()
17:50:03.072 -> D NimBLERemoteService: << NimBLERemoteService(): 0x1801
17:50:03.072 -> D NimBLEClient: Service Discovered >> status: 0 handle: 16
17:50:03.072 -> D NimBLERemoteService: >> NimBLERemoteService()
17:50:03.106 -> D NimBLERemoteService: << NimBLERemoteService(): 0xfe95
17:50:03.138 -> D NimBLEClient: Service Discovered >> status: 0 handle: 35
17:50:03.171 -> D NimBLERemoteService: >> NimBLERemoteService()
17:50:03.171 -> D NimBLERemoteService: << NimBLERemoteService(): 0xfef5
17:50:03.303 -> D NimBLEClient: Service Discovered >> status: 0 handle: 49
17:50:03.303 -> D NimBLERemoteService: >> NimBLERemoteService()
17:50:03.303 -> D NimBLERemoteService: << NimBLERemoteService(): 00001204-0000-1000-8000-00805f9b34fb
17:50:03.566 -> D NimBLEClient: Service Discovered >> status: 0 handle: 58
17:50:03.566 -> D NimBLERemoteService: >> NimBLERemoteService()
17:50:03.566 -> D NimBLERemoteService: << NimBLERemoteService(): 00001206-0000-1000-8000-00805f9b34fb
17:50:03.764 -> D NimBLEClient: Service Discovered >> status: 14 handle: -1
17:50:03.764 -> D NimBLEClient: << Service Discovered
17:50:03.764 -> D NimBLEClient: << retrieveServices
17:50:03.764 -> D NimBLERemoteService: >> retrieveCharacteristics() for service: 0x1800
17:50:03.993 -> D NimBLERemoteService: Characteristic Discovered >> status: 0 handle: 3
17:50:04.025 -> D NimBLERemoteCharacteristic: >> NimBLERemoteCharacteristic()
17:50:04.025 -> D NimBLERemoteCharacteristic: << NimBLERemoteCharacteristic(): 0x2a00
17:50:04.025 -> D NimBLERemoteService: Characteristic Discovered >> status: 0 handle: 5
17:50:04.025 -> D NimBLERemoteCharacteristic: >> NimBLERemoteCharacteristic()
17:50:04.025 -> D NimBLERemoteCharacteristic: << NimBLERemoteCharacteristic(): 0x2a01
17:50:04.025 -> D NimBLERemoteService: Characteristic Discovered >> status: 0 handle: 7
17:50:04.059 -> D NimBLERemoteCharacteristic: >> NimBLERemoteCharacteristic()
17:50:04.059 -> D NimBLERemoteCharacteristic: << NimBLERemoteCharacteristic(): 0x2a02
17:50:04.256 -> D NimBLERemoteService: Characteristic Discovered >> status: 0 handle: 9
17:50:04.256 -> D NimBLERemoteCharacteristic: >> NimBLERemoteCharacteristic()
17:50:04.256 -> D NimBLERemoteCharacteristic: << NimBLERemoteCharacteristic(): 0x2a04
17:50:04.354 -> D NimBLERemoteService: Characteristic Discovered >> status: 14 handle: -1
17:50:04.354 -> D NimBLERemoteService: << Characteristic Discovered
17:50:04.354 -> D NimBLERemoteService: << retrieveCharacteristics()
17:50:04.354 -> D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a00
17:50:04.386 -> D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a01
17:50:04.386 -> D NimBLERemoteCharacteristic: >> retrieveDescriptors() for characteristic: 0x2a02
[... 180 lines cut ...]
17:50:09.805 -> D NimBLERemoteCharacteristic: Got 7 bytes
17:50:09.805 -> I NimBLERemoteCharacteristic: Read complete; status=14 conn_handle=0
17:50:09.805 -> D NimBLERemoteCharacteristic: << readValue length: 7 rc=0
17:50:09.838 -> -- Battery: 72
17:50:09.838 -> - get history control
17:50:09.838 -> D NimBLERemoteService: >> getCharacteristic: uuid: 00001a10-0000-1000-8000-00805f9b34fb
17:50:09.838 -> D NimBLERemoteService: >> retrieveCharacteristics() for service: 00001204-0000-1000-8000-00805f9b34fb
17:50:10.463 -> D NimBLERemoteService: Characteristic Discovered >> status: 14 handle: -1
17:50:10.463 -> D NimBLERemoteService: << Characteristic Discovered
17:50:10.463 -> D NimBLERemoteService: << retrieveCharacteristics()
17:50:10.463 -> D NimBLERemoteService: >> retrieveCharacteristics() for service: 00001204-0000-1000-8000-00805f9b34fb
[... cut ...]
17:50:11.847 -> D NimBLEClient: Got Client event 
17:50:11.847 -> D NimBLEClient: Notify Recieved for handle: 33
17:50:11.847 -> D NimBLEClient: checking service 0xfe95 for handle: 33
17:50:11.879 -> D NimBLEClient: Got Notification for characteristic Characteristic: uuid: 0x1001, handle: 33 0x0021, props:  0x10
17:50:11.879 -> Descriptor: uuid: 0x2902, handle: 34
17:50:12.871 -> D NimBLEClient: Got Client event 
17:50:12.871 -> I NimBLEClient: disconnect; reason=531, 
17:50:12.871 -> D NimBLEClientCallbacks: onDisconnect: default
```

h2zero, 2024-06-04:
This looks to be an issue of multiple same UUID's. The solution is to provide an index parameter to `getCharateristic()`. Please try this branch https://github.com/h2zero/NimBLE-Arduino/tree/remote-attr-index

h2zero, 2026-03-20:
Closing this as stale

closed 2026-03-20 by h2zero