h2zero/NimBLE-Arduino #522: Pair everytime on Android and Connect fail on IOS

CharlesXSong, 2023-03-31:
I' m using NimBLE in ESP32(server), the SecurityAuth is:
~~~c
NimBLEDevice::setSecurityAuth(BLE_SM_PAIR_AUTHREQ_BOND | BLE_SM_PAIR_AUTHREQ_SC);
~~~
ISSUE:
---
After several mobile phones(client) connected to the same ESP32, the connection fails on the **IOS** with the following error message: 
`Peer removed pairing information`

And need to pair everytime on **Android**.

---
How to do with that?

h2zero, 2023-03-31:
How many devices are connecting to this? The default max bond count is 3, seems you need to increase this.

CharlesXSong, 2023-03-31:
> How many devices are connecting to this? The default max bond count is 3, seems you need to increase this.

Sorry, I didn't make it clear. The max bond count I set is 1:
~~~C
#define CONFIG_BT_NIMBLE_MAX_CONNECTIONS 1
~~~

---
The situation is this device(ESP32) only connects to ONE phone at a time, and several phones will connect to it frequently for testing.

CharlesXSong, 2023-03-31:
> Sorry, I didn't make it clear. The max bond count I set is 1:
>
> ```c
> #define CONFIG_BT_NIMBLE_MAX_CONNECTIONS 1
> ```
>
> The situation is this device(ESP32) only connects to ONE phone at a time, and [...]

Oh NO! I completely confused connection and bond count o(╥﹏╥)o... @h2zero Sorry for wasting ur valuable time.
I rechecked the `nimconfig.h`, found that CONFIG_BT_NIMBLE_MAX_BONDS is commented, means the default value  **3** is defined :
~~~C
/** @brief Un-comment to change the number of devices allowed to store/bond with */
// #define CONFIG_BT_NIMBLE_MAX_BONDS 3

#ifndef CONFIG_BT_NIMBLE_MAX_BONDS
#define CONFIG_BT_NIMBLE_MAX_BONDS 3
#endif
~~~

---
Consider the limited RAM, is there any way to not allow device  to limit the bond number? (If this can solve the pair issue)

h2zero, 2024-06-13:
I think this is resolved in the latest commits in release 1.4 and master.

Calling `NimBLEDevice::setSecurityInitKey(3);` in older commits should also work;

closed 2024-06-13 by h2zero