h2zero/NimBLE-Arduino #528: Pairing key is not requested upon connect but only when accessing secure characteristic

nelfata, 2023-04-16:
Is it possible to know if this is a feature or a bug. Trying to establish a connection between ESP32 and NRFConnect (iPhone), the passkey is only requested to be entered when accessing the characteristic. 
Typically, it should be prompted upon establishing a connection. 

    NimBLEDevice::setSecurityAuth(true, true, true);
    NimBLEDevice::setSecurityPasskey(pin);
    NimBLEDevice::setSecurityIOCap(BLE_HS_IO_DISPLAY_ONLY);

Please advise. Thanks.

h2zero, 2023-04-16:
This is how BLE works, unlike bluetooth classic. Here we can have devices that have some data secured and other data not secure. What you can do if you want to force a secure connection is call `NimBLEDevice::startSecurity` from the onConnect callback.

nelfata, 2023-04-16:
Ok that's great, that's what I was looking for. Thank you.

chegewara, 2023-04-17:
Im not sure, but with bluedroid version it is possible to force pin pairing on connection with security option.
One of this with specific option, but i dont remember which one:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/security/SampleClient_authentication_passkey.cpp#L72
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/security/SampleClient_authentication_passkey.cpp#L68

chegewara, 2023-04-17:
Security is not client nor server side, its GAP, so its "device side".

nelfata, 2023-04-18:
Sorry, I am no expert in BLE. I see that the security is based on the characteristic settings and not on the service nor the device. I am trying to have the server not allow a connection to be established until the PIN is entered by the client (as an example).

h2zero, 2024-06-04:
> Sorry, I am no expert in BLE. I see that the security is based on the characteristic settings and not on the service nor the device. I am trying to have the server not allow a connection to be established until the PIN is entered by the client (as an example).

This was answered earlier, in the `onConnect` callback you need to call `NimBLEDevice::startSecurity`, the connection ID is provided by the callback parameter.

closed 2024-06-04 by h2zero