h2zero/NimBLE-Arduino #536: Are there examples of how to run tasks on the Nimble stack core?

jackjansen, 2023-04-29:
In the usage tips document it is stated that it is best to use the library from a task that runs on the Nimble stack core in stead of the normal core.

An example of how to do this would be good. But actually a pointer to code that does this would also be helpful already...

jackjansen, 2023-05-15:
Ah, thanks, that's a very useful repo! A few remarks:
- I have the feeling that the `BLE_scan` example you're referring to has a bug: it seems like `pBLE->start()` isn't called (or maybe I'm overlooking it)
- `NimBLE_Client` seems like a better example (it does quite a bit more in the task, and also shows one way of communicating between the tasks, using a shared variable)
- Maybe add a reference to this repo in the documentation?

h2zero, 2023-05-15:
That example is in the master branch which has some API changes, the blocking scan call no longer uses `start()` instead the `getResults()` call replaces it.

Yes the Client example is probably better but for simplicity I used the scan example.

There is a reference in the README, perhaps it should be made more apparent though.

closed 2024-06-12 by h2zero