h2zero/NimBLE-Arduino #537: Notification/Indication calback with content containing "0d"

Fredovsky, 2023-04-30:
A BLE device sends through indications or notifications a message of the following type :
`76 40 0d 33 43 33 34 36 34 20`

Because the message contains "0d", the callback parameter length is wrong (2 instead of 10 in this example), and I am unable to display the full message using :
```
void notifyCB(NimBLERemoteCharacteristic* pRemoteCharacteristic, uint8_t* pData, size_t length, bool isNotify){
    std::string str = "Message = " + std::string((char*)pData, length);
    Serial.println(str.c_str());
}
```

I am using the code from the NimBLE_Client example.

Any help appreciated.

Fredovsky, 2023-05-01:
Issue fixed using the following :

```
for(uint8_t i = 0; i<length; i++)
	{
		Serial.print(" ");
		Serial.printf("%02x", pData[i]); // Prints 2 digits hex value
		
	}
```

Variable `length` is correct, just the `std::string((char*)pData, length)` returns a truncated string if it contains `0d`

closed 2023-05-01 by Fredovsky

chegewara, 2023-05-03:
Actually this is the problematic line
`Serial.println(str.c_str());`