h2zero/NimBLE-Arduino #542: Reconnect after Board reboot

mr-deamon, 2023-05-06:
Hi there
I now tried to get this to run a full afternoon :)

I'll try to get my ESP32C3 (no classic BT, this might have been easier?!) act as a ble-server (with security) and have an android phone connect to it. This connection should be automatically re-established, as soon they come close enough. If i switch off and on the phones BT it works (reconnect), if i reboot the ESP it doesn't. Android keeps listing the server in "saved devices" but not in the "connected" list.

My goal: i have a button which should only work, if a previously joined device (my phone) is nearby. no data exchange, nothing, just "if successfully authorized and connect, forward the key-press. if not, do nothing.

any ideas? would greatly appreciate it!
chris

h2zero, 2023-05-14:
Sounds like you need to enable bonding on the esp32. 

Here is an example that uses it:
https://github.com/h2zero/NimBLE-Arduino/blob/remote-attr-index/examples/NimBLE_Secure_Server/NimBLE_Secure_Server.ino

mr-deamon, 2023-06-11:
Thank you. It's finally working on android with this code:
https://gist.github.com/mr-deamon/b1c8b82cfefffac89e0f75b08a7292ee

But ios does not show the pin-prompt. I tried 1000 things, but it remains the same, ios simply connects (without bonding, strangely). Any ideas why? This should be working on android and ios the same way...
Thanks
Chris

JanB97, 2023-07-03:
For me this doesn't work at all.
The ESP show one bound device after reconnecting but I am completely unable to connect again unless I remove the esp from my phone and pair again.
I am using android. Is there anything I am missing?
I habe tried the secure server example and the code from @mr-deamon .

pmmcmullen94, 2023-07-10:
> For me this doesn't work at all. The ESP show one bound device after reconnecting but I am completely unable to connect again unless I remove the esp from my phone and pair again. I am using android. Is there anything I am missing? I tried the secure server example and the code from @mr-deamon .

I'm having the same issue. In windows, it connects fine but then cycles between "Paired" "Connected" "Paired" "Connected" and I have to completely forget the device and readd. For my iPhone, it'll connect fine the first time but then I have to forget the device to readd it.

I'm using auth and whitelisting. It requires a pin code to be entered to start up

h2zero, 2024-06-12:
its been some time for this, sorry for the lack of response. 

try: `NimBLEDevice::setSecurityInitKey(3);` and let me know if it helps.

pmmcmullen94, 2024-06-12:
> its been some time for this, sorry for the lack of response.
> 
> try: `NimBLEDevice::setSecurityInitKey(3);` and let me know if it helps.

Appreciate the reply! Your timing is perfect actually haha. Projects been on pause for the past year but I’m picking it up again in about a week. I forgot I posted this and it’s one of the issues I have to test soon. 

I’ll give that a shot and let you know how it goes. Thanks!