h2zero/NimBLE-Arduino #543: Device name is getting truncated when setting custom advertising data.

ankgt, 2023-05-14:
I want to first thank the author for developing and maintaining such a fantastic library. Not only is the library amazing, his support and helpfulness on the forum is something I have not seen anywhere. Kudos!

I have been experimenting with setting custom manufacturer data in the advertising packet. I have written the following code and it works great with the exception that the device name is getting truncated to 6 characters (shows only as NimBLE). Now from my research I believe the advertising data length should not exceed 31 bytes. I am ensuring this (27 bytes as per my calculation including the LEN and TYPE fields).

However I find that if I manually set the device name again on (by doing pAdvertising->setName) the name appears correctly (NimBLE-Arduino).

Is this a bug or something I am doing wrong? Thanks!

`

#include <NimBLEDevice.h>

static NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
static uint32_t count = 60000;

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  
  NimBLEDevice::init("NimBLE-Arduino");
}

void loop() {
  // Stop the advertising.
  pAdvertising->stop();

  // Prepare the manufacturer data.
  uint8_t mfgData[6];
  // Set the mfg code to silicon labs 0x02FF (for testing).
  mfgData[0] = 0xFF;
  mfgData[1] = 0x02;
  // Set the next three bytes for the count.
  uint8_t* ptr = (uint8_t*) &count;
  mfgData[2] = ptr[0];
  mfgData[3] = ptr[1];
  mfgData[4] = ptr[2];
  mfgData[5] = ptr[3];
  pAdvertising->setManufacturerData(std::string((char*)&mfgData, 6));
  
  // Set the name again. If I dont do this, the name is truncated in the advertising data!
  // pAdvertising->setName("NimBLE-Arduino");
  
  // Restart the advertising.
  pAdvertising->start();
  
  Serial.printf("Advertising count = %d\n", count);
  count++;
  delay(5000);
}
`

h2zero, 2023-05-14:
Hello, the reason you are seeing this is because you have created the advertising instance before initializing the stack here:
```
static NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
```
To correct this I suggest changing the line above to:
```
static NimBLEAdvertising *pAdvertising = nullptr;
```
The in your setup you would add this line after `init()`:
```
pAdvertising = NimBLEDevice::getAdvertising();
```

ankgt, 2023-05-15:
Thanks very much, that did the trick! 

Incidentally I had used the 'NimBLE_Service_Data_Advertiser' example as the base. That also suffers from this issue. Shall I make a pull request for this?

h2zero, 2023-05-15:
Great, sure please submit a PR for this, thanks!

closed 2023-05-16 by ankgt