h2zero/NimBLE-Arduino #547: getting duplicate service

sdetweil, 2023-05-22:
followup on previous add/removeServiceUUID issue

service  created only once, is the only advertised service in nrfConnect.

connect to see services available, but this recently created service is shown twice, both empty( no characteristics)

connecting app says characteristics missing

[image]

fails on the test app sample I provided last week.

I had seen this when I did addServiceUUID() multiple times, but I have verified there is only one add call done now

while in loop()

stop advertising (prior service UUID)
remove prior advertising UUID ( now none)
call library which creates service on server
lib creates characteristics (6)
lib starts service
lib returns service address
add serviceuuid (added debug in nimble method to report count of UUIDs, after add call, only 1)
start advertising

mobile scan(nrfConnect)
correct UUID advertised
connect
services as shown in screen shot

h2zero, 2023-05-22:
To resolve this I had to move this:
```
  // disconnect the(any) connected clients
  for (auto it: pServer->getPeerDevices()) {
    pServer->disconnect(it);
  }
  while (pServer->getConnectedCount() > 0) {
    yield();
  }
```
to line 466.

Also removed this line in the OTA lib https://github.com/sdetweil/ArduinoBleOTA/blob/6cf3edbd73d72b94848ef9c62ae5592160ee2400/src/ArduinoBleOtaClassNimBle.cpp#L74

And finally added:
```
pServer->advertiseOnDisconnect(false);
```
here https://github.com/sdetweil/testapp/blob/ce7302b5a1942b65d1dd7824e1cf8e6e66e8b607/src/testapp.cpp#L521

There were simply too many things triggering service start at once, also if the scanner is active then the stack will fail to load the services correctly because it is handling the results and blocking the mutex.

sdetweil, 2023-05-22:
thanks

1. i had moved up the client disconnect, as there is a disconnect handler that wants to turn advertising back on. 
1.5 will add the delay
3. I had removed the service->start(), but it didn't affect the results. 
4. >pServer->advertiseOnDisconnect(false);     I had assumed that was the default..  cause I manually restart on disconnect (see 1!)..
5. scanner...  i have changed it to cycle less often, 2 seconds every 10. to save power... we shall see

sdetweil, 2023-05-22:
ok, not there yet

now I have one service,  but empty

also turned off the scanner when we are starting the OTA service onward. 

in loop check for OTAService requested first,  (start service and set Running
then check if OTA Running (yes), skip scan

i don't restart advertising in onDisconnect handler if OTARunning 
```
void startOTAService()
{
  OTARunning = 2;
#ifdef ESP32
  advertising = false;
  #ifdef USE_NIMBLE
    Sprintln("stop advertising");
    pAdvertising->stop();
    Sprintf("removing advertised UUID=");
    Sprintln(activeService[status]->getUUID().toString().c_str());
    pAdvertising->removeServiceUUID(activeService[status]->getUUID());
  #else
    pAdvertising->stop();
  #endif
  // disconnect the(any) connected clients
  disconnectClient();  // the code you shared + the delay(100)
  // start the Over The Air update service
  OTAService = ArduinoBleOTA.begin(pServer, InternalStorage, HW_NAME_INFO.c_str(), HW_VER, SW_NAME.c_str(), SW_VER);
  OTARunning = 1;
  ArduinoBleOTA.setUploadCallbacks(*(new myUploadCallbacks()));
  Sprintln("restart advertising after OTA Service setup");
  Sprintln("setting OTA service as only");
  #if USE_NIMBLE
     pAdvertising->addServiceUUID(OTAService->getUUID());
     Sprint("added UUID to advertising, should be only=");
     Sprintln(OTAService->getUUID().toString().c_str());
     pAdvertising->start();
     Sprintln("restart advertising");
  #else
    pAdvertising->setServiceUUID(OTAService->getUUID());
    pAdvertising->start();
  #endif
    advertising = true;
#else

#endif

}
```
//if characteristic write handler 
write characteristic=9A64
OTA Start requested

// later ...1 second called from loop

//start OtaService
stop advertising
removing advertised UUID=00000001-27b9-42f0-82aa-2e951747bbf9
disconnected (force disconnect)
restart advertising after OTA Service setup
setting OTA service as only
added UUID to advertising, should be only=15c155ca-36c5-11ed-adc0-9741d6a72f04
restart advertising
connected  // nrfConnect 
on disconnect start advertising= 15c155ca-36c5-11ed-adc0-9741d6a72f04  //restart advet on ota service
disconnected // nrfConnect end of onDisconnect() handler

h2zero, 2023-05-24:
I left comments on the latest commit

sdetweil, 2023-05-24:
thank you. left comments back, really on disconnect

sdetweil, 2023-05-25:
is there a usecase that service->start() is appropriate?

I don't own the ota service library

sdetweil, 2023-05-25:
service->start().. thanks..  is this a wasted call, if already running? or a damaging call?  

again , I don't own the library, don't "want" to own my own fork...

h2zero, 2023-05-25:
It's problematic to call it the way it's being done here. if you don't want to use a fork then I would suggest adding the service at the startup and just remove it until needed.

sdetweil, 2023-05-31:
I updated the test app and the library to remove the pService->start()

but see  the service in  nrfconnect after connecting to the device..
shouldn't be there if I removed it..

h2zero, 2023-08-13:
This works as intended by adding `pServer->start();` after `ArduinoBleOTA.begin`

closed 2024-04-17 by sdetweil