h2zero/NimBLE-Arduino #549: scanCompleteCB not called after scan completes

jackjansen, 2023-05-24:
I have a situation where the `scanCompleteCB` I passed to `scanner->start()` isn't always called.

I have a list of BLE servers that I keep by name. After boot, I periodically scan with a duration until I have the MAC address for each server.

I'm using tasks for connecting to servers, and the "main task" (Arduino loop) handles starting and stopping the scanner if any of my expected servers isn't known yet. But in some situations the scanner doesn't call the `scanCompleteCB` when it stops scanning. I _think_ this happens if I connect to a server in a connection task at roughly the same time as I start the scan (but I'm not 100% sure of this).

If I get rid of the tasks and do the connecting/communicating/disconnecting in the main arduino `loop()` the problem goes away and the scan complete callback is always called.

h2zero, 2023-05-25:
>But in some situations the scanner doesn't call the scanCompleteCB when it stops scanning. I think this happens if I connect to a server in a connection task at roughly the same time as I start the scan (but I'm not 100% sure of this).

Sounds correct to me, if you are initiating a connection at the same time as starting the scan, the scan start will fail, you would need to check the return value of `scan->start()` to confirm.

jackjansen, 2023-05-25:
Ah, thanks! That should have been obvious.

closed 2023-05-25 by jackjansen