h2zero/NimBLE-Arduino #555: ESP32 (any) + whitelisting working example??

sanastasiou, 2023-06-07:
Hi everyone, I have tried countless configurations, hacks, scoured the interwebs and every possible example of code I could think of.

I just cannot get whitelisted bonded devices to work with setScanFilter and ESP32 C3 device.. 

I have found several open bugs regarding this matter, so I am tending to believe this functionality is not available for these devices.

That said, if anyone has a working example of whitelisted devices I would be happy to compensate you for your efforts.

Best Regards

h2zero, 2023-06-24:
Are the devices you're bonded with using a random address type?

sanastasiou, 2023-06-24:
Yes, android and iPhone devices. I guess there is a bug in the ESP32 backend which means this functionality does not work at all right?

h2zero, 2023-07-04:
Yes, the esp32 controller does not support whitelisting random addresses

closed 2023-07-04 by sanastasiou

h2zero, 2023-07-04:
Any of the Nordic BLE chips support this and can be used with this library.