h2zero/NimBLE-Arduino #559: getName returns empty string

rodneygmorgan, 2023-06-28:
What am I missing or doing wrong?

--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Found 8 devices
Connected to 
Length of getName() : 0
Found our UUID : 4fafc201-1fb5-459e-8fcc-c5c9c331914b

BLE server code=========
#include <Arduino.h>
#include "NimBLEDevice.h"

NimBLEServer *pServer;

// void setup()
void setup(void)
{
    Serial.begin(115200);
    Serial.println("Starting NimBLE Server");
    
    NimBLEDevice::init("Ranger Beacon");

    // Create Server
    pServer = NimBLEDevice::createServer();

    // Create Service
    NimBLEService* pService = pServer->createService("4fafc201-1fb5-459e-8fcc-c5c9c331914b");

    // Create Characteristic
    NimBLECharacteristic* pCharacteristic = pService->createCharacteristic("beb5483e-36e1-4688-b7f5-ea07361b26a8");

    NimBLEDescriptor* pDscr = pCharacteristic->createDescriptor("Ranger",NIMBLE_PROPERTY::READ | NIMBLE_PROPERTY::WRITE,20);
    pDscr->setValue("at Barn");
    
    // Start the service
    pService->start();

    // Set Characteristic value
    pCharacteristic->setValue("Barn");

    NimBLEAdvertising* pAdvertising = NimBLEDevice::getAdvertising();
    pAdvertising->addServiceUUID(pService->getUUID()); 

    // Start advertising
    pAdvertising->setScanResponse(true);
    pAdvertising->start(); 

    Serial.println("Advertising Started");

}

void loop()
{
    delay(2000);
    Serial.print(".");
}

BLE Client code=================
#include "NimBLEDevice.h"
#include <Arduino.h>

void setup() {
  char foo[48];
  Serial.begin(115200);
  Serial.println("Starting NimBLE Client");

  NimBLEDevice::init("");
    
  NimBLEScan *pScan = NimBLEDevice::getScan();
  NimBLEScanResults results = pScan->start(10);
    
  NimBLEUUID serviceUuid("4fafc201-1fb5-459e-8fcc-c5c9c331914b");

  Serial.printf("Found %i devices\n",results.getCount());
  for(int i = 0; i < results.getCount(); i++) {
    NimBLEAdvertisedDevice device = results.getDevice(i);
        
    if (device.isAdvertisingService(serviceUuid)) {
      Serial.printf("Connected to %s\n",device.getName().c_str());
      Serial.printf("Length of getName() : %i\n",strlen(device.getName().c_str()));
      Serial.printf("Found our UUID : %s\n",device.getServiceUUID().toString().c_str());
      NimBLEClient *pClient = NimBLEDevice::createClient();
          
      if (pClient->connect(&device)) {
        NimBLERemoteService *pService = pClient->getService(serviceUuid);
                
        if (pService != nullptr) {
          NimBLERemoteCharacteristic *pCharacteristic = pService->getCharacteristic("beb5483e-36e1-4688-b7f5-ea07361b26a8");
                    
          if (pCharacteristic != nullptr) {
            std::string value = pCharacteristic->readValue();
            //Serial.printf("Characteristic value %s\n",value);
            // print or do whatever you need with the value
          }
        }
      } else {
      // failed to connect
      }
            
      NimBLEDevice::deleteClient(pClient);
    }
  }
}

void loop() {
  // put your main code here, to run repeatedly:
}

h2zero, 2023-06-29:
The device may not be advertising it's name so there is no data.

rodneygmorgan, 2023-06-29:
I based the server code off the new user guide and included that code in my
request for help.  The documentation indicates the value specified in the
init() command is used as the device name.  Is there a different
place/method I should use to supply an "advertised" name?  Thanks for the
quick reply.

h2zero, 2023-06-29:
Yes you need to set the name in the advertisement, the device name in init only sets the name value for the characteristic.

closed 2024-06-04 by h2zero