h2zero/NimBLE-Arduino #562: NimBLECharacteristic::setValue() is calculating the wrong message length.

orbian, 2023-06-30:
I am programming and ESP32 with platform.io.

NimBLECharacteristic::setValue(const char *s) is setting the attribute size to the entire size of the buffer, not just the '\0' terminated string.

For example this does the correct thing and sends 22 bytes to the client (string data plus null terminator):
>  uint16_t cap;
>     std::string value;
>
>     outgoingCharateristic->setValue("12345678901234567890A");
>     cap = outgoingCharateristic->getValue().capacity();
>     printf("Capacity: %d\n", cap);
>     value = outgoingCharateristic->getValue();
>     printf("%s\n", value.c_str());

The server prints out:

> Capacity: 22
> 12345678901234567890A

That's what the client receives as well.

But if I declare a char buffer and fill it in, what gets sent is the entire buffer, no matter how large it is.

>     char str[45];
>     uint16_t cap;
>     std::string value;
> 
>     sprintf(str, "12345678901234567890A");
>     outgoingCharateristic->setValue(str);
>     cap = outgoingCharateristic->getValue().capacity();
>     printf("Capacity: %d\n", cap);
>     value = outgoingCharateristic->getValue();
>     printf("%s\n", value.c_str());

In this case the server prints:

> Capacity: 45
> 12345678901234567890A

My client receives a 45 byte buffer and thinks there is more data than there really is. It's still the null terminated string but followed with garbage.

Changing the buffer to char str[1500] causes and error.
>
>E NimBLEAttValue: value exceeds max, len=1500, max=512
>

setValue() uses strlen() to determine the length of the data so it shouldn't give a value that exceeds the null termination. If I strlen() the string before passing it to setValue() it does report the correct value. However once inside setValue(), only the buffer length is returned.

h2zero, 2023-07-04:
This may be an issue of the template used when compiling. I would suggest casting the char array to a uint8_t pointer and passing the strlen value to setvalue directly.

h2zero, 2025-06-17:
@hobzcalvin this is probably a bit of a bug if you have this issue, however, you can simply set the value with the string without calling the c_str() method.