h2zero/NimBLE-Arduino #565: how to reference Nimble-Arduino in a "Arduino as an ESP-IDF component" project?

mhaberler, 2023-07-03:
I have a [Platformio project ](https://github.com/mhaberler/flowsensor/tree/sync-sdkconfig)with an ESP32 using Arduino framework, and Nimble-Arduino

This is set up as a "esp-idf mit Arduino als component"  as described here: https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/esp-idf_component.html 

I include Nimble-Arduino in [platformio.ini](https://github.com/mhaberler/flowsensor/blob/main/platformio.ini#L84)

when building, I get a significant number of "symbol redefined" warnings, like so:

````
In file included from .pio/libdeps/m5stack_coreS3/NimBLE-Arduino/src/NimBLEAddress.cpp:14:
.pio/libdeps/m5stack_coreS3/NimBLE-Arduino/src/nimconfig.h:234: warning: "CONFIG_BT_NIMBLE_ACL_BUF_COUNT" redefined
 #define CONFIG_BT_NIMBLE_ACL_BUF_COUNT 12
 
In file included from .pio/libdeps/m5stack_coreS3/NimBLE-Arduino/src/nimconfig.h:4,
                 from .pio/libdeps/m5stack_coreS3/NimBLE-Arduino/src/NimBLEAddress.cpp:14:
.pio/build/m5stack_coreS3/config/sdkconfig.h:132: note: this is the location of the previous definition
 #define CONFIG_BT_NIMBLE_ACL_BUF_COUNT 20
 
Compiling .pio/build/m5stack_coreS3/lib905/NimBLE-Arduino/NimBLEBeacon.cpp.o
In file included from .pio/libdeps/m5stack_coreS3/NimBLE-Arduino/src/NimBLEAdvertisedDevice.cpp:15:
.pio/libdeps/m5stack_coreS3/NimBLE-Arduino/src/nimconfig.h:154: warning: "CONFIG_BT_NIMBLE_ROLE_CENTRAL" redefined
 #define CONFIG_BT_NIMBLE_ROLE_CENTRAL
````
obviously these come from collisions between nimconfig.h and the [sdkconfig.<env>fragment](https://github.com/mhaberler/flowsensor/blob/sync-sdkconfig/sdkconfig.m5stack_coreS3)

how should I deal with this?

ignore?
adjust all options in sdkconfig.m5stack_coreS3 to match the values from nimconfig.h?

the full build log is [here](https://github.com/mhaberler/flowsensor/blob/sync-sdkconfig/logs/buildlog.txt)
the esp-idf version is 4.4.5.230614

thanks in advance
Michael

h2zero, 2023-07-04:
I suggest using this instead: https://github.com/h2zero/esp-nimble-cpp

mhaberler, 2023-07-04:
thanks, will do that and report - leaving open until then

mhaberler, 2023-07-04:
added esp-nimble-cpp as git submodule under lib/
adjusted platformio.ini 
worked right away - thanks a lot!

-Michael
reference: https://github.com/mhaberler/flowsensor/tree/main

closed 2023-07-04 by mhaberler