h2zero/NimBLE-Arduino #566: BLE HID Battery Level Indicator

ThomasAtBBTF, 2023-07-05:
I am experiencing the same problem which was discussed here with NimBLE-Arduino:

https://github.com/espressif/arduino-esp32/pull/6864/files/35fb47ebebc352e43f935c0bf70beb6b0e3629ea 

I tried to add NimBLE2902 but this is not implemented here..

Can someone please take a look?

JustTryingToGetSomeWorkDone, 2023-10-31:
NimBLE2902 is documented (well the api anyway) here: https://h2zero.github.io/esp-nimble-cpp/class_nim_b_l_e2904.html

I'm afraid I can't help anymore than this. I'm not even sure what 2902 refers to or why it's special among descriptors/characteristics. Your link doesn't really describe a situation as far as I can understand. It's just a code edit. The only discussion is about the indentation, and I'm sure you don't mean that.

NimBLE has the best library documentation among all the BLE libraries I've tried, but it's still not perfect. An example of each function would go a long way towards making it easier for newbies like me to understand, and a short description of what it does and when it's needed would make it perfect. Maybe one day I'll make it my life goal to improve library documentations. Until then...

Good luck!

h2zero, 2023-11-02:
Oops missed this. But for anyone else finding it there is no 2902 descriptor class because it's function is handled by the stack/library.

ignis32, 2024-05-12:
Not sure if it is about the same issue I had, but I had battery indicator updated only if my esp is deleted and paired again on Win 11.
Comment from  here solved my issue:

 

 #https://github.com/T-vK/ESP32-BLE-Keyboard/issues/210#issuecomment-1995165739
 
 
 However, I am running version 3.1 of the library for now.

h2zero, 2024-06-04:
I think this would work in the application code `NimBLEHIDDevice::batteryLevel()->notify` after setting the level.

closed 2024-06-13 by h2zero