h2zero/NimBLE-Arduino #567: Reconnecting not working (in Android)

JanB97, 2023-07-06:
So I already posted to one issue here I think, but I am not sure anymore if it is the same problem.
I am trying for days now to get my android phone to reconnect to the ESP32, after it or the phone were rebooted or got out of range from another.
I tried with different service UUIDs and Characteristics and everything I could find here in the issues.

I now enabled verbose logging and the weird thing is, I can connect exactly one time to the esp, thats working (kind of) fine and I can see the connect in the log.
If I now reboot the ESP or the Phone, I don't even see a connection try in the log. 
Even if I hit connect on the phone, nothing, neither the phone nor the ESP show any sign of trying to connect to each other.

Now I though, well maybe it's the phone (Nothing Phone 1), so I tried with the old phone (OnePlus 7t) and .... exactly the same thing.
Am I missing something here? Can someone with knowledge about this help?
I am currently playing around with the Server and Secure Server examples, but because it basically never changed I cannot really provide anymore code.

Can someone tell me what I would need to do with the code examples to make it work?

As a side note:
No I wouldn't really need to connect to the ESP for my usecase, but because of Androids battery management it is way easier to autostart you app with the Broadcast receiver of device connect instead of scanning for yourself, and it is way more efficient.

JanB97, 2023-07-07:
Okay, so I tried looking at the logcat of the phone and I get the following when trying to connect manually:
2023-07-07 19:48:21.836 20931-20931 CachedBluetoothDevice   com.android.settings                 D  No profiles. Maybe we will connect later for device

It sounds pretty straight forward, but how to add a profile?
Nothing I tried worked

h2zero, 2023-07-09:
What app are you using on the android device? Does it work with nRFConnect?

JanB97, 2023-07-09:
Yeah I can connect with nRFC and I can read the characteristics, however even after bonding with the app, the esp doesn't automatically reconnect to the phone even in bond state. This means the action "ACTION_ACL_CONNECTED" is never fired.
I kind of got it to work with using the NIMBLEHIDDEVICE and emulating a mouse (with a keyboard my phone would popup a keyboard config notification) but than I can't really read the other characteristics anymore.
I am now trying to fix this on the phones site, maybe I can somehow bond my phone to a custom profile - haven't really fully checked the ble api as I first thought fixing this on the esp would be easier.
Otherwise I would have to run a service that scans ble constantly, but that would kill the battery.

JanB97, 2023-07-10:
I think this is fixable with the CompanionDeviceManager in Android at least.
I will try that route and most likely close the issue shortly, unless it turns out to not fix the problem.

JanB97, 2023-07-22:
I am pretty sure now, that my companion device service does work and detects the ble presence.
I have only tested with android 13 and it limits definitely to device running android 12 and up, but that's more or less a android limitation, so I will close this issue now.

closed 2023-07-22 by JanB97