h2zero/NimBLE-Arduino #568: Stuck with limited length of LocalName

rainerfritz, 2023-07-09:
Hi!

I am developing an App which connects to my ESP32 devices from Android and iOS. On Android I can only use the local name. The local name advertised seems to be limited to 8 characters. I am aware of the limitations in length when using BLE, but thought it could be longer than 8. I would need at least 10 characters. So i wanted to use the Manufacturer Data to have a workaround for this situation. Do you have a working example with the latest release of the library? Can't find one. Is there any way to expand the local name? The device name is advertised as expected, so thats not the problem. Is there a full example on how to implement custom advertising? Any help would be much appreciated.

h2zero, 2023-07-09:
This depends on how much data you are advertising. If there is not enough space the name will be shortened.

rainerfritz, 2023-07-09:
The device name is very long with up to 17 characters. Do you mean reducing this would lead in more space for the local name? Apart from that we have the usual UUID for the device ID.

Koxx3, 2023-08-10:
I have the same issue but it's weird, because on IOS, it works fine, but on Android (Pixel 6A), it's limited to 8 characters.
So, I don't think it's purely related to the advertising datas.

any idea ?

rainerfritz, 2023-08-10:
iOS caches the device name when it connects to the device and shows then always the full cached device name, which is not the case on android. I also use nordic nrf52 based RAK modules and use the bluefruit library where we don't have that problem. In addition to my case I have the problem that my capacitor community-le library can not proper read Scan Response messages, so I am stuck to use manufacturer data on Android, with the drawback that in the bluetooth devices I have only 8 character names. I tried a lot in reducing characters for the local and device name, but nothing worked so far, even when I only use 10 char long name it gets shortened.

h2zero, 2023-08-12:
If you would like to post some example code it would be easier for me to answer this.

rainerfritz, 2023-08-14:
Thats the code for advertising:

```
NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
    pAdvertising->reset();
    pAdvertising->setName(strBLEName);  //BLE Local Name
    pAdvertising->setManufacturerData(strBLEManufData);
    pAdvertising->addServiceUUID(SERVICE_UUID);
    pAdvertising->setScanResponse(false);    // true ANDROID  false IPhone
    pAdvertising->start(0);
```
Even when I set the name to 1234567890 I get only 8 characters. This also happens when I don't use manufacturer data.

Koxx3, 2023-12-11:
any news about this issue ?

h2zero, 2023-12-12:
No news, the service uuid I believe is the cause here. It is taking up likely 18 bytes out of 28 and the name would be limited to 8 because of the 2 bytes for the data type and length.

rainerfritz, 2023-12-12:
The only way is to use a scan-response or to use manufacturer data. Currently I go with the latter. It would be really awesome to get an example code here to implement correctly a scan-response message.

closed 2024-06-05 by h2zero

kei969777, 2024-12-27:
any news about this issue ?

h2zero, 2025-01-02:
Not sure what news you are expecting. If the advertised name is shortened then it is due to there not being enough room for it in the advertisement packet.

If you want more control over the advertisement packet and scan response just create 2 instances of `NimBLEAdvertisementData` and pass them as arguments to `setAdvertisementData` and `setScanResponseData`.

kei969777, 2025-01-10:
https://github.com/h2zero/NimBLE-Arduino/blob/7f2c2ba3a030024fa7e05fbe81d33ced88f80f5e/examples/BLE_EddystoneTLM_Beacon/BLE_EddystoneTLM_Beacon.ino#L50