h2zero/NimBLE-Arduino #572: Cannot pair to sony camera but works when using Bluedroid

Lemin2, 2023-07-24:
Hi h2zero,

Hello, I encountered a problem with connecting esp32 and Sony ZV-1 camera, and I would like to ask you for advice.

I modified the example program of Nimble-Arduino library a bit, and I wanted to use esp32 to control my ZV-1 camera, but when esp32 and the camera paired, the program got stuck at `D NimBLEClient: >> secureConnection()`, and then reported an error `E NimBLEClient: secureConnection: failed rc=13`.

I know that ZV-1 would pop up a confirm pairing window, when it's in pairing mode, and the connected device has sent any requests to it. I can successfully simulate this process with the nrf connect app on Android.

I saw on GitHub that someone was also testing how to use Nimble or Bluedroid to control Sony cameras. I tried his program and found that, using nimble-arduino would also get stuck at `D NimBLEClient: >> secureConnection()`, and disconnect after 10 seconds; 
However, using Bluedroid can pair and control normally. And after pairing successfully with Bluedroid, I changed to flash the Nimble-Arduino program,and this time esp32 could also pair with the camera, but the device name displayed on the camera was Bluedroid program's.

I tried changing the settings of `NimBLEDevice::setSecurityAuth()` or `NimBLEDevice::setSecurityIOCap()`, but it didn't work.  Am I doing anything wrong or is there a bug in the Nimble-Arduino library?

Thank you for your time and help.

h2zero, 2023-07-24:
All I can see in the log is this:
```
secureConnection: failed rc=13
```
Which is 

0x0d | BLE_HS_ETIMEOUT | Operation timed out.
-- | -- | --

This may be due to the configuration of the security settings in `NimBLEDevice::setSecurityIOCap()` and `NimBLEDevice::setSecurityAuth()`.

Lemin2, 2023-07-24:
> All I can see in the log is this:
> 
> ```
> secureConnection: failed rc=13
> ```
> 
> Which is
> 
> 0x0d	BLE_HS_ETIMEOUT	Operation timed out.
> This may be due to the configuration of the security settings in `NimBLEDevice::setSecurityIOCap()` and `NimBLEDevice::setSecurityAuth()`.

Hi h2zero,
It seems that the camera only requires `no io capacity` and `bonding` to be functional.
so I've tried to set my esp32 with:
```
NimBLEDevice::setSecurityIOCap(BLE_HS_IO_NO_INPUT_OUTPUT);
NimBLEDevice::setSecurityAuth(BLE_SM_PAIR_AUTHREQ_BOND |  BLE_SM_PAIR_AUTHREQ_SC);
```
but this doesn't help. I'm still getting the same error. I also tried other combinations and they don't help too. 
Then I found another one [using NRF52840 the chip and bluefruit library to control a sony camera](https://[github.com](https://github.com/tao-j/alpha-ble)/tao-j/alpha-ble). seems in the code he is able to manually issue an SMP Pairing Request or an SMP Security Request. I don't know how to achieve this with an esp32 so I'm just assuming ble libraries would do this automatically. so I just tried to duplicate the default GAP security parameters of the bluefruit library:
```
// default is Just Work
static const ble_gap_sec_params_t _sec_param_default =
{
  .bond         = 1,
  .mitm         = 0,
  .lesc         = LESC_SUPPORTED,
  .keypress     = 0,
  .io_caps      = BLE_GAP_IO_CAPS_NONE,
  .oob          = 0,
  .min_key_size = 7,
  .max_key_size = 16,
  .kdist_own    = { .enc = 1, .id = 1},
  .kdist_peer   = { .enc = 1, .id = 1}
};
```
And I'm doing like this:
```
  ble_hs_cfg.sm_bonding=1;
  ble_hs_cfg.sm_mitm=0;
  ble_hs_cfg.sm_sc=1;
  ble_hs_cfg.sm_keypress=0;
  ble_hs_cfg.sm_io_cap=BLE_SM_IO_CAP_NO_IO;
  ble_hs_cfg.sm_oob_data_flag=0;
  ble_hs_cfg.sm_our_key_dist=BLE_SM_PAIR_KEY_DIST_ENC|BLE_SM_PAIR_KEY_DIST_ID;
  ble_hs_cfg.sm_their_key_dist=BLE_SM_PAIR_KEY_DIST_ENC|BLE_SM_PAIR_KEY_DIST_ID;
```
Unfortunately this doesn't make a difference either. I'm still getting the same error.
I'm still new to ble, and don't know why Nimble and Bluedroid behave differently, although they seem to be using similar interfaces. the Bluedroid code just sets encryption level with `BLEDevice::setEncryptionLevel(ESP_BLE_SEC_ENCRYPT);` and all the thing goes flawlessly; but I just don't know what kind of magic is happening here, and what to do to get Nimble working like that. Could you please give me some more advice on that? 
Best reguards

h2zero, 2023-08-12:
I think this may have something to do with `BLE_SM_PAIR_AUTHREQ_SC` or `ble_hs_cfg.sm_sc=1;` being enabled. Try removing this, by setting `ble_hs_cfg.sm_sc=0;`

h2zero, 2024-06-05:
@Lemin2  Any updates?

Lemin2, 2026-01-05:
@h2zero should we make these settings to  be same as bluedroid's default settings to address this issue?

Lemin2, 2026-01-06:
if you already have a working idf project with nimble, [esp-nimble-cpp](https://github.com/h2zero/esp-nimble-cpp)and arduino things settled, then you can try to find these two configs in menuconfig, or you can just edit the sdkproject file which locates in the project's root directory, use search to find these settings and set them correctly.

finfinack, 2026-01-07:
Thanks for the pointer. I didn't observe any changes when setting it (but to be honest, I would have been surprised as my problem is with pairing, not with scanning). I'll keep looking at ESP-IDF Bluedroid and see if I can get it to work there and try to see whether I can see anything that sticks out. Also happy to share my PoCs in case you'd like to reproduce it. FWIW, I'm trying to connect to a Sony A7V or A7IV but my assumption was that it shouldn't matter that much which Sony it is.

finfinack, 2026-01-08:
I did that before indeed as well. It didn't have any effect, i.e. I saw the same errors on the ESP logging side (`secureConnection: failed rc=13`) but no change in behavior. Apologies - I don't want to waste your time here, I have way too little time to properly dig into it because it's just a side project among 2 others. If you're happy to lazily go back and forth, I'd be more than happy to but I'm also not offended if you just tell me to come back better informed later :)

h2zero, 2026-01-11:
Thank you @finfinack, I'll look into this in depth this week, maybe I can find something interesting.