h2zero/NimBLE-Arduino #574: Accessing characteristics takes too long

jojo0329, 2023-07-24:
Hi！
I debugged with nrf connect debugging software and found that the phone connected esp32 takes about 3s from the start of the connection to the end of accessing the features. Can this time be reduced and what can be done about it?
Can you point me how to solve this problem?
Thank you!

[image]
[image]

This is a part of my code, I use freertos multitasking：

```
#include <NimBLEDevice.h>

BLECharacteristic * pTxCharacteristic;
static BLEServer* pServer;
#define SERVICE_UUID        "6E400001-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_RX "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"
#define CHARACTERISTIC_UUID_TX "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"
#define DEVICE_NAME "AZ50011Z02G237240001"
bool isAdvertising = true;  
int clientCount = 0;    

void addManufacturerData(NimBLEAdvertisementData& advertisementData,const std::string &data,unsigned char type) 
{
    char cdata[2];
    cdata[0] = data.length() + 1;
    cdata[1] = type ;  // 0xff
    advertisementData.addData(std::string(cdata, 2) + data);
} // addManufacturerData

class MyServerCallbacks: public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
        Serial.printf("Client connected");
        Serial.printf("Multi-connect support: start advertising");
        isAdvertising=false;
        clientCount++;
        // NimBLEDevice::startAdvertising();
    };

    void onConnect(BLEServer* pServer, ble_gap_conn_desc* desc) {
        Serial.printf("Client address: ");
        Serial.printf("%s",NimBLEAddress(desc->peer_ota_addr).toString().c_str());
        pServer->updateConnParams(desc->conn_handle, 24, 48, 0, 60);
    };
    void onDisconnect(BLEServer* pServer) {
        Serial.printf("Client disconnected - start advertising");
        isAdvertising=false;
        clientCount--;
    };

    void onMTUChange(uint16_t MTU, ble_gap_conn_desc* desc) {
        Serial.printf("MTU updated: %u for connection ID: %u\n", MTU, desc->conn_handle);
    };
};

class MyCharacteristicCallbacks: public BLECharacteristicCallbacks {
    void onWrite(BLECharacteristic *pCharacteristic) {
      std::string value = pCharacteristic->getValue();

      if (value.length() > 0) 
      {
        parseData();//BLEprotocolData

      }
    }
};
void setup()
{
  BLEDevice::init(DEVICE_NAME);
  // ADDED to support larger MTU
  NimBLEDevice::setMTU(512);//test

  // Create the BLE Server
  pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(SERVICE_UUID);
  // Create a BLE Characteristic
  pTxCharacteristic = pService->createCharacteristic(
                                        CHARACTERISTIC_UUID_TX,
                                        NIMBLE_PROPERTY::NOTIFY
                                       );
  BLECharacteristic *pRxCharacteristic = pService->createCharacteristic(
                                        CHARACTERISTIC_UUID_RX,
                                        NIMBLE_PROPERTY::WRITE_NR
                                       );

  pRxCharacteristic->setCallbacks(new MyCharacteristicCallbacks());
  pServer->setCallbacks(new MyServerCallbacks());

  pService->start();

  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(SERVICE_UUID);

  uint8_t Adv_DATA[] = {0x01, 0x00, 0x01, 0x03};
  BLEAdvertisementData oAdvertisementCustom = BLEAdvertisementData();
  BLEAdvertisementData oScanResponseData = BLEAdvertisementData();
  oAdvertisementCustom.setFlags(0x06);
  pAdvertising->setAdvertisementData(oAdvertisementCustom);
  pAdvertising->setScanResponse(true);
  addManufacturerData(oScanResponseData,std::string((char *)&Adv_DATA[0], sizeof(Adv_DATA)),0xFF);
  addManufacturerData(oScanResponseData,std::string(DEVICE_NAME),0x09);
  pAdvertising->setScanResponseData(oScanResponseData);

  #ifdef ESP_PLATFORM
      NimBLEDevice::setPower(ESP_PWR_LVL_P9); /** +9db */
  #else
      NimBLEDevice::setPower(9); /** +9db */
  #endif

  // Start advertising
  BLEDevice::startAdvertising(); 
}

void loop()
{
	while (1)
	{
    if(BLEDevice::getInitialized() && !isAdvertising && clientCount<10)
    {
      BLEDevice::startAdvertising(); 
      isAdvertising = true;
    }
    delay(50);
	}

}
```

h2zero, 2023-08-12:
Hello, not much can be done about this from the server side. The connection parameters are set by the client/phone and that will be the biggest factor in speed.

h2zero, 2024-06-05:
Closing as stale.

closed 2024-06-05 by h2zero