h2zero/NimBLE-Arduino #576: Does not write data to the characteristic without first reading

Casper17, 2023-07-27:
Hi!
I am trying to write data to the characteristic using the example of NimBLE_Client to reduce the time for connecting and writing data to the characteristic, I commented out reading from the characteristic and now the recording does not pass, an error pops up, but the result of the recording returns the truth!
[image]
[image]

JustTryingToGetSomeWorkDone, 2023-10-31:
Your log shows that after writing your data to 0xbeef, it skips to the string "Done with this device!" That's waaaaaaaay down the nested if statements, all the way at the bottom of the first if statement on line 202... that means you messed up your braces "{ }" somewhere. If you haven't already fixed it and still want to fix it, check your braces. I find clicking on an opening "{" or closing "}" brace will highlight the complementary brace on the other side. Make sure they open and close the correct "if" blocks. All sorts of strange things can happen if your if statements get out of alignment. Things get skipped that you thought would run, and things that you didn't think would run are suddenly spitting out serial output and making disconnections/notifications.

If I'm misreading the situation, I apologize and mean no insult. It's just a common mistake to make, and I don't know anything about your situation beyond what's in your post.

Good luck.

h2zero, 2024-06-05:
@JustTryingToGetSomeWorkDone got this i'd say 👍

closed 2024-06-05 by h2zero