h2zero/NimBLE-Arduino #582: Impossible to reconnect after some time and multiple connections

Koxx3, 2023-08-14:
Hello,

this is a bug that I have been tracking for a long time.
It's difficult to reproduce.

After some time, it becomes impossible to reconnect to the ESP BLE.
In 99.9% of the time, it works fine.
But sometime, it's locked in a bad state.

I finally managed to capture the log.

After that it is impossible to reconnect.
I never seen the 'onAuthenticationComplete'.

I found a hack ... but it's really dirty.
I have to deinit NimBLE and reinit.
then it works again.

It's a complicated project, it's difficult to extract the BLE code.

Koxx3, 2023-08-14:
Here is the log :
[https://koxx3.fr.eu.org:8086/ble_conn_ko.txt](url)

Note that at line 4704, I decide to reset the NimBLE (after 5 failure).

I've tried to connect another phone, it's not the phone or the phone BLE stack.
When stuck. I tried with android and iphone, impossible to reconnect.
They are stuck when "reading services" or "connect" state.
Tried with my app or BlueLighing app. Same thing.

Koxx3, 2023-08-14:
As you can see, My ESP act as a BLE server, it also constinuously advertise and scan for devices.

No memory issue : 
MEM left = 54256        idle = 5584     tft = 1936      wifi = 3020     btn = 2452      canrx = 1600    cantx = 1884    sound = 408     serial = 4500   btCntr = 1580   btHost = 5212   CAN_WD_BI = 1544        CAN_LORX = 3524        Tmr = 1388      ipc0 = 256      esp_timer = 2816        loopTask = 5264

Koxx3, 2023-08-14:
Here is a the important code extraction : 
[https://koxx3.fr.eu.org:8086/ble_conn_ko.cpp](https://koxx3.fr.eu.org:8086/ble_conn_ko.cpp)

Koxx3, 2023-08-14:
@h2zero , any help would be greatly appreciated.
As said, it's very difficult to capture and I don't see anything wrong the my current code.

Koxx3, 2023-08-14:
Also, my PIO config : 

```
[env]
platform = https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.5/platform-espressif32-v.2.0.5.zip
framework = arduino, espidf
board = esp32dev

lib_deps = 
	nimble=https://github.com/h2zero/NimBLE-Arduino.git@1.4.1+sha.bc333cc
```

h2zero, 2023-08-14:
Hi @Koxx3, I don't see any obvious problem in the logs. I suspect an issue with the BLE controller as it's well known to hang, especially in the version of the arduino core/IDF you are on.

Unfortunately, if that's the case, what you are doing to workaround it is the best way to resolve the issue.

Koxx3, 2023-08-14:
thanks a lot for your feedback.
do you known which version of arduino/idf could improve this sitation ?
migration on a large project is always a pain. If I can target a version, it will help :D

h2zero, 2023-08-14:
The old versions, such as 1.0.6 were better in this regard, but I doubt you want to go back to that. I believe this should be fixed when the new version based on IDF 5.x is released, you could try testing it now.

h2zero, 2024-06-05:
@Koxx3 are you still having this issue?

h2zero, 2024-06-13:
@Koxx3 A number of commits where just made to the master branch and 1 to the release branch that should resolve this.

h2zero, 2024-11-20:
Closing this as there has been no response, please re-open if still and issue.

closed 2024-11-20 by h2zero