h2zero/NimBLE-Arduino #585: change address and payload

beckmx, 2023-09-08:
hello there, i was reading the docs and studying the code, I was wondering if it is possible to define a custom Bluetooth address for the device and then advertise a custom payload, at least the payload part from what i can see is somehow achievable with the beacon example, thanx a lot for this lib!! :)

h2zero, 2023-09-24:
Sorry for the late response. Yes, a custom MAC address is able to be set with calls to the esp controller:
```
uint8_t new_mac[8] = {0x01,0x02,0x03,0x04,0x05,0x06};
esp_base_mac_addr_set(new_mac);
```
I'm not sure what you mean by a custom payload but most anything is possible here.

beckmx, 2023-09-24:
And should it work the same for the nrf 51?

Koxx3, 2024-09-11:
hello.
I also need to change address periodically to advertise rotating MACs.
esp_base_mac_addr_set doesn't seem to be taken into account dynamically.
is there any solution ?

h2zero, 2024-09-15:
For nRF and others you can use this:
```
ble_addr_t addr;
ble_hs_id_gen_rnd(0, &addr); // Set first arg to 1 if you want NRPA
ble_hs_id_set_rnd(addr.val);
```

closed 2026-03-17 by h2zero