h2zero/NimBLE-Arduino #586: Low Sleep Mode

j45p41, 2023-09-19:
Hi,

I am trying to use put NRF52840 to sleep.

I have tried Arduino_nRF5x_lowPower library but it seems to not work with thislinrary showing the following error:

.pio/libdeps/nrf52840_dk_adafruit/Arduino nRF5x lowPower/src/Arduino_nRF5x_lowPower.h:25:10: fatal error: nrf_sdm.h

I simply want to put the device to sleep and wake up on interrupt, while using ADXL362 and interrupt source.

Otherwise great library that I want to keep on using.

Thanks!

h2zero, 2023-09-24:
This is not a function of this library, more of a core function. That said you can follow the Nordic examples and use `NRF_POWER->SYSTEMOFF = 1;` to enter system off sleep mode after enabling a GPIO wakeup.

h2zero, 2023-11-17:
Just to add to my response above, the  Arduino_nRF5x_lowPower library is only for use with the softdevice, not NimBLE. n-able core has the `systemPowerOff()` function that should work for this purpose.

closed 2024-06-05 by h2zero