h2zero/NimBLE-Arduino #588: Simple GATT Security "Just Works" style configuration

truedat101, 2023-09-20:
Hello - I am excited to find this code available on Arduino.  Thank you for the library.  For my ESP32, I am attempting to set up a simple secure GATT BLE bonding with the "Just Works" pairing method, between two devices with no screens and no passkey to be exchanged.  

I reviewed description of the various security models available from Bluedroid:

https://github.com/espressif/esp-idf/blob/master/examples/bluetooth/bluedroid/ble/gatt_security_server/tutorial/Gatt_Security_Server_Example_Walkthrough.md 

And the new user guide: https://h2zero.github.io/NimBLE-Arduino/md__new_user_guide.html

I have modified the example NIMBLE_Secure_Server to 

```
NimBLEDevice::init("Halo");
#ifdef ESP_PLATFORM
    NimBLEDevice::setPower(ESP_PWR_LVL_P9); /** +9db */
#else
    NimBLEDevice::setPower(9); /** +9db */
#endif

  NimBLEDevice::setSecurityAuth(true, true, true);
  // NimBLEDevice::setSecurityPasskey(0);
  // NimBLEDevice::setSecurityPasskey(123456);
  NimBLEDevice::setSecurityIOCap(BLE_HS_IO_NO_INPUT_OUTPUT);
  NimBLEServer *pServer = NimBLEDevice::createServer();
  NimBLEService *pService = pServer->createService("MyService");

```

My question is, what is the correct way to set up "Just Works" style pairing method.  My partner device that is going to bond with this ESP32 is using this style of authentication.

h2zero, 2023-09-24:
Hello, the only thing you need to do for this is to enable bonding:
```
NimBLEDevice::setSecurityAuth(true, false, true);
```

"Just works" pairing is the default pairing mode so nothing special needs to be done, so only bonding would need to be enabled.

truedat101, 2023-09-26:
Bueno, thank you for clarifying!