h2zero/NimBLE-Arduino #596: EXTNEDED Legacy connection works on ESP32-S3 but not on C3

j45p41, 2023-10-20:
I am running the exact same code to connect to BLE app on mobile which works on Esp32-S3 but always disconnects on ESp32-C3.

Do I need to change any settings for switching to C3? Thanks in advance here is the log:

`I NimBLEDevice: BLE Host Task Started
I NimBLEDevice: NimBle host synced.
D NimBLEServer: >> createService - 4fafc201-1fb5-459e-8fcc-c5c9c331914b
D NimBLEServer: << createService
D NimBLEService: >> start(): Starting service: UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0xffff
D NimBLEService: Adding 1 characteristics for service UUID: 4fafc201-1fb5-459e-8fcc-c5c9c331914b, handle: 0xffff
D NimBLEService: << start()
1
2
3
primary service
           uuid 0x1800
         handle 1
     end_handle 5
characteristic
           uuid 0x2a00
     def_handle 2
     val_handle 3
   min_key_size 0
          flags [READ]
characteristic
           uuid 0x2a01
     def_handle 4
     val_handle 5
   min_key_size 0
          flags [READ]
primary service
           uuid 0x1801
         handle 6
     end_handle 9
characteristic
           uuid 0x2a05
     def_handle 7
     val_handle 8
   min_key_size 0
          flags [INDICATE]
ccc descriptor
           uuid 0x2902
         handle 9
   min_key_size 0
          flags [READ|WRITE]
primary service
           uuid 4fafc201-1fb5-459e-8fcc-c5c9c331914b
         handle 10
     end_handle 13
characteristic
           uuid beb5483e-36e1-4688-b7f5-ea07361b26a8
     def_handle 11
     val_handle 12
   min_key_size 0
          flags [READ|WRITE|NOTIFY|INDICATE]
ccc descriptor
           uuid 0x2902
         handle 13
   min_key_size 0
          flags [READ|WRITE]
D NimBLEExtAdvertising: >> Extended Advertising start
D NimBLEExtAdvertising: << Extended Advertising start
Started advertising
D NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_CONNECT 
D NimBLEServerCallbacks: onConnect(): Default
Client connected:: 4d:c6:91:05:52:74
D NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_ADV_COMPLETE
Advertising instance 1 stopped
 client connecting
D NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_PHY_UPDATE_COMPLETE
D NimBLEServer: << handleGATTServerEvent
D NimBLEServer: >> handleGapEvent: BLE_GAP_EVENT_DISCONNECT
Client disconnected
D NimBLEServerCallbacks: onDisconnect(): Default
`

JustTryingToGetSomeWorkDone, 2023-10-31:
The logs are not very helpful at finding the exact problem, but they look like they're pointing to your onConnect or onAdvertisingComplete callbacks (I think that's what it's called). I'm currently working on esp32-c3s as well, and I've got mine successfully staying connected to an Android app I built for testing. Could you post that part of your code and maybe I can help?

Another possibility is the app might have something going wrong. What are you using to connect to the c3? Is it a custom app or something like nRF Connect? I also have no trouble connecting to nRF Connect with my c3s, so I recommend testing with that app if you're not already doing so.

h2zero, 2023-10-31:
I don't see anything in the logs that point to the issue. More verbose logs will be needed if you are able.

j45p41, 2023-11-03:
Hi both,

first of all thank you very much indeed for replying to my message.

@JustTryingToGetSomeWorkDone  thanks for pointing out that you have it working with the C3. My troubleshooting steps should have been to run the code in isolation, which I obviously didn't do and thee fact that you confimed it woke in the C3 prompted me to do this.

To explain I have moved over from S3 to C3 and the same code worked on the S3. I now know that the reason for the failure was due to running a WiFi Scan in STA mode before launching BLE and, although this seemed to work on the S3, for some (good reason probably) does not run on the C3.

I have been able to get the desired result by storing the values from the WiFi scan in SPIFFS and then doing an ESP.restart() and carrying on. I am sure there must be a way to do both simultaneously, but after doing a WiFi.mode(WIFI_OFF); I made some progress but could not keep the client connected so just did it the way I described.

Now all I need is to get an answer to my question on using the [NRF528 in CODED_PHY](https://github.com/h2zero/NimBLE-Arduino/issues/581) and I can start sending the devices for user testing. Thanks again for making the library and humbly helping out.

I will now close this issue.

closed 2023-11-03 by j45p41