h2zero/NimBLE-Arduino #599: NimBLEDevice::setPower(int dbm) not working as expected (NRF52840)

lilindian16, 2023-11-02:
**NRF52840-QIAA**
n-able core version = 1.0.0
NimBLE-Arduino version = 1.4.1

platformio.ini config:
_platform = https://github.com/h2zero/platform-n-able.git#1.0.0
framework = arduino
lib_deps = h2zero/NimBLE-Arduino@^1.4.0
upload_protocol = jlink
board = generic_nrf52840
debug_tool = jlink_

Using NimBLEDevice::setPower(int dbm) seems to not change the tx power from what I can tell.

Iterating through the following dbm values shows no change using LightBlue to read RSSI:
`{RADIO_TXPOWER_TXPOWER_Neg4dBm, RADIO_TXPOWER_TXPOWER_0dBm, RADIO_TXPOWER_TXPOWER_Pos3dBm, RADIO_TXPOWER_TXPOWER_Pos4dBm}`
Reading RSSI on LightBlue app shows no change between all four settings

The workaround I have used so far is changing nimconfig `MYNEWT_VAL_BLE_LL_TX_PWR_DBM=0` to `MYNEWT_VAL_BLE_LL_TX_PWR_DBM=4` with platformio build flag `build_flags = -DMYNEWT_VAL_BLE_LL_TX_PWR_DBM=4`
Reading RSSI on LightBlue shows immediate change after device is flashed with new config

h2zero, 2023-11-03:
Interesting, do you see any change if you use the numerical values instead of the macros, i.e -4 etc?

j45p41, 2024-02-23:
> **NRF52840-QIAA** n-able core version = 1.0.0 NimBLE-Arduino version = 1.4.1
>
> platformio.ini config: _platform = https://github.com/h2zero/platform-n-able.git#1.0.0 framework = arduino lib_deps = [...]

hi @lilindian16 I used your suggestion and it increased power to 4dbm - did you ever get it to 8dbm? if so can you share pls?

h2zero, 2024-07-02:
Fixed in #679

PR #679 (2024-06-24): Nimble core v1.5.0 
Update NimBLE core to v1.5.0

fixes #599
closed 2024-07-02 by h2zero