h2zero/NimBLE-Arduino #601: NimBLEDevice::setDeviceName doesn't work

wwj718, 2023-11-05:
from the changelog: [setDeviceName allowing for changing the device name while the BLE stack is active.](https://github.com/h2zero/NimBLE-Arduino/blob/bc333ccb6e8d9ff2059af9cbd5006a290a4de3a5/CHANGELOG.md?plain=1#L17)

setDeviceName  not working properly

I use esp32(esp32doit-devkit-v1, NimBLE-Arduino@^1.4.1), below is my code:

The DeviceName is always "test1"

```ino
#include <Arduino.h>
#include <NimBLEDevice.h>
#include <NimBLEServer.h>
#include <NimBLEAdvertisedDevice.h>

/*
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps = h2zero/NimBLE-Arduino@^1.4.1
*/

static NimBLEAdvertising *pAdvertising = nullptr;
static uint32_t count = 0;
BLEUUID serviceUUID("123456-0003-4538-BCD7-7ECFB51297C1");

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");
  NimBLEDevice::init("test1");
  pAdvertising = NimBLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(serviceUUID);
}

void loop() {
  pAdvertising->stop();
  pAdvertising->start();
  delay(3000);
  // setDeviceName allowing for changing the device name while the BLE stack is active.: https://github.com/h2zero/NimBLE-Arduino/blob/bc333ccb6e8d9ff2059af9cbd5006a290a4de3a5/CHANGELOG.md?plain=1#L17
  NimBLEDevice::setDeviceName("test2"); // do not work
}
```

h2zero, 2023-11-05:
If you are meaning to change the advertised name you can only do that while advertising is stopped. That name change you are using here will change the device name characteristic value only in this case.

h2zero, 2023-11-06:
There are 2 names, one in the advertisement and the other in the device name characteristic. If you just want to change the advertised name you should use  `pAdvertising->setName("new name");`.

wwj718, 2023-11-06:
It works for me, thanks a lot!

closed 2023-11-06 by wwj718