h2zero/NimBLE-Arduino #606: Support in Arduino 3.x.x

rvbc1, 2023-11-20:
Hello, I am aware that the Arduino 3.X.X release is currently in the alpha phase, but I am developing a new product based on the ESP32H2, so I need to use this version of Arduino because the previous version doesn't support the H2 MCU. I've tested the classic BLE from the Arduino built-in libraries and it seems to be working fine. My project is just starting and I suppose that the Arduino release of the new 3.X.X with support for the H2 module will be ready by the time my project is finishing. However, I want to use the NimBLE-Arduino library because I find it really great and I have a previous project (for the ESP32) based on this, so it would be easier for me to transfer the written modules to the new project than to rewrite them with the BLE from Arduino libraries. But currently, when I try to compile the Arduino alpha release with NimBLE-Arduino, I get a lot of errors like the one above:

.pio/libdeps/local/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:457:4: error: 'esp_timer_handle_t' does not name a type; did you mean 'TimerHandle_t'?

It seems that some structures or includes have changed in the IDF, so my question is: are there any plans to add compatibility for this library with IDF 5.1?

Here is my current configuration for reproduction in PIO. I can also add main.cpp with a sample, but any use of NimBLE-Arduino results in errors in the newest Arduino release

[platformio]
default_envs = local

[env]
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
board = esp32-h2-devkitm-1
framework = arduino

build_unflags = 
	-std=gnu++11
	-std=gnu99

build_flags = 
	-D ESP32
	-std=gnu++2a
	-I include

lib_deps = 
	https://github.com/h2zero/NimBLE-Arduino/archive/refs/tags/1.4.1.zip
	

Thank you for any information. Are there plans to add support for this, and if so, when is it roughly scheduled to be implemented? If there are no plans for this support, I would also appreciate that information so that I do not wait unnecessarily and can begin rewriting the modules immediately. Once again, thank you in advance for your response and for this library, as I have found it to be excellent to use (even if further development is not planned, for the ESP32, it's the best BLE library available).

h2zero, 2023-11-20:
Hello, you're welcome and thank you for the kind words.

I am hoping to be able to update this soon but it will take some time as it's quite a large job to update the nimble core files.

mhaberler, 2023-11-23:
I was able to work around this temporarily by using

`lib_deps =
	https://github.com/h2zero/esp-nimble-cpp.git
`

h2zero, 2023-11-24:
Had a look at this today, @mhaberler has the correct solution in this case.

The cause of the issue is the use of `platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5` since that config enables NimBLE in the Arduino lib builder and therefore conflicts with the NimBLE core files here. So the easy way to fix this is to use https://github.com/h2zero/esp-nimble-cpp

Jason2866, 2024-06-09:
We had this https://github.com/h2zero/NimBLE-Arduino/issues/645 and other issues in our alpha builds of core 3.0.0. And we wanted to support of the new chips.
This is easy possible to get with the great library `esp-nimble-cpp` :-)
So we changed the needed settings in sdkconfig to make the compiled Arduino libs compatible to this lib. It did work in first attempt!

The Tasmota Arduino framework will always (try to) have the settings to be compatible with `esp-nimble-cpp`

h2zero, 2024-06-13:
Closing this as support for Arduino 3.x is established. Support for C6, C2, H2 and others may not be possible and are a separate issue.

closed 2024-06-13 by h2zero