h2zero/NimBLE-Arduino #607: Unable to Send Messages to ESP32 using "BLE_Write" or "Server"

AlEnterni, 2023-11-22:
Hello

 I am currently facing an issue with my ESP32 development board and I'm seeking assistance to resolve it.

I have been trying to use the "BLE_Write" and "Server" examples for my ESP32, but unfortunately, I am unable to successfully send messages to the ESP32. Strangely, I can read messages without any issues, but the writing functionality seems to be problematic. I use mobile apps like nRF Connect or BLE Scanner.
I have followed the examples and documentation carefully, but I haven't been able to figure out what's causing the problem. I would greatly appreciate it if someone could provide guidance on how to troubleshoot and fix this issue.

Thank you in advance for your assistance. I'm looking forward to resolving this issue and continuing my ESP32 development journey.

Best regards,

h2zero, 2024-06-05:
@AlEnterni Could you provide some example code to reproduce this, or better yet, some logging with debug logs enabled?

h2zero, 2024-06-09:
This may have been a case of needing to write with or without response depending on the situation. Closing this now as there has been no response from the author.

closed 2024-06-09 by h2zero