h2zero/NimBLE-Arduino #612: NomBLE and BLE are not currently compatible.

zawatse, 2023-11-30:
Hello, Im using this library to create a custom game controller. Im also using the BLE lib in the same project to allow the controller to function as a keyboard. I need to implement a workaround currently, because NimBLE-Arduino\src\NimBLEHIDDevice.h and BLE\src\BLEHIDDevice.h use the same symbol in the #ifndef, _BLEHIDDEVICE_H_. My suggestion is to change this symbol in the NimBLE-Arduino lib to _NIMBLEHIDDEVICE_H_, or something else, to allow these libs to be compatible. I tried to push this change to the repo in a branch, but I dont have the right permissions.

h2zero, 2023-11-30:
This is interesting, I'm not understanding why you need to use both libraries. Is there something missing in NimBLE that has been overlooked?

zawatse, 2023-12-03:
Our use case evolved over time. It's possible we could switch to using NimBLE for both the keyboard and gamepad "modes" for our device, but our code originally only used the BLE lib, and we added use of NimBLE later for the gamepad mode. In your readme you mention that "this library is intended to be compatible with the original ESP32 BLE functions," so I was thinking this could be a useful change (regardless of if it is still necessary for our case), but just a suggestion.

h2zero, 2024-12-09:
Resolved in [5ade97a](https://github.com/h2zero/NimBLE-Arduino/commit/5ade97a13dfad1e4e74b290f4a276bed7d1771f4)

closed 2024-12-09 by h2zero