h2zero/NimBLE-Arduino #617: pScan->start(...) blocking forever

Dario-Ciceri, 2023-12-15:
Hi, when I run pScan->start(5, scanEndedCB, false); or pScan->start(5, false); even after it has finished scanning the program goes into infinite loop and I don't understand why, my code doesn't include cycles after the call to start (apart from the loop of course) and it just seems like the program is waiting for something and doesn't even crash, I hope you can provide me with a solution, thanks in advance.

Dario-Ciceri, 2023-12-15:
maybe I found the problem, the scan doesn't stop even after the endscancb is called

Dario-Ciceri, 2023-12-15:
nope, even by adding in the callback NimBLEDevice::getScan()->stop(); it doesn't work and keep waiting for something
```
16:15:50.979 -> Scan Ended
16:15:50.979 -> D NimBLEScan: >> stop()
16:15:50.979 -> D NimBLEScan: << stop()
```

Dario-Ciceri, 2023-12-15:
this https://github.com/h2zero/NimBLE-Arduino/issues/297 helps but it's not perfect btw

h2zero, 2023-12-15:
What MCU is this on? Can you share a code snippet to reproduce?

Dario-Ciceri, 2023-12-17:
> What MCU is this on? Can you share a code snippet to reproduce?

I'm using an esp32, I fixed it by using a task on core 0.

What I don't understand is why both the blocking and non-blocking versions, after scanning for "x" seconds, don't return control to the main loop, obviously with two cores there are no problems for the application I need to run however I would like to be able to solve or at least understand if I am doing something wrong.

Part of the code:
```
//(NOT USED) quick fix blocking scan https://github.com/h2zero/NimBLE-Arduino/issues/297
#include <NimBLEDevice.h>
#include <vector>
using namespace std;

TaskHandle_t scanTask;
bool scanTaskRunning = false;

#define ServerName "VR-PARK"

NimBLEAddress ServerAddress = 0x48472e390db3;

void scanEndedCB(NimBLEScanResults results);

// UUID HID
static NimBLEUUID serviceUUID("1812");
// UUID Report Charcteristic
static NimBLEUUID charUUID("2a4d");

static NimBLEAdvertisedDevice *advDevice;

static bool doConnect = false;
static bool clientConnected = false;
static uint32_t scanTime = 0; /** 0 = scan forever */

class ClientCallbacks : public NimBLEClientCallbacks {
    void onConnect(NimBLEClient *pClient) {
      Serial.println("Connected");
      clientConnected = true;
      /** After connection we should change the parameters if we don't need fast response times.
            These settings are 150ms interval, 0 latency, 450ms timout.
            Timeout should be a multiple of the interval, minimum is 100ms.
            I find a multiple of 3-5 * the interval works best for quick response/reconnect.
            Min interval: 120 * 1.25ms = 150, Max interval: 120 * 1.25ms = 150, 0 latency, 60 * 10ms = 600ms timeout
      */
      pClient->updateConnParams(120, 120, 0, 1);
    };

    void onDisconnect(NimBLEClient *pClient) {
      Serial.print(pClient->getPeerAddress().toString().c_str());
      Serial.println(" Disconnected - Starting scan");
      clientConnected = false;
      //NimBLEDevice::getScan()->start(scanTime, false);
    };

    /** Called when the peripheral requests a change to the connection parameters.
          Return true to accept and apply them or false to reject and keep
          the currently used parameters. Default will return true.
    */
    bool onConnParamsUpdateRequest(NimBLEClient *pClient, const ble_gap_upd_params *params) {
      if (params->itvl_min < 24) { /** 1.25ms units */
        return false;
      } else if (params->itvl_max > 40) { /** 1.25ms units */
        return false;
      } else if (params->latency > 2) { /** Number of intervals allowed to skip */
        return false;
      } else if (params->supervision_timeout > 100) { /** 10ms units */
        return false;
      }

      return true;
    };
[... 295 lines cut ...]
      doConnect = true;
      if (!scanTaskRunning) {
        vTaskResume(scanTask);
        scanTaskRunning = true;
      }
    }
  } else if (!clientConnected && !scanTaskRunning) {
    Serial.println("Not connected to bluetooth remote, scanning...");
      vTaskResume(scanTask);
      scanTaskRunning = true;
  }
  return false;
}

bool isBluetoothConnected() {
  return clientConnected;
}
```

setup():
`Serial.print("Starting BLE Services");
  if (!BLESetup()) {
    Serial.print("Can't start BLE Services!");
    while (1) delay(10);
  }`

h2zero, 2023-12-29:
A quick look suggests that the scan task is being suspended in various situations but never resumed so the scan is never restarted.

h2zero, 2024-06-05:
Closing this as it does not appear to be an issue with the library.

closed 2024-06-05 by h2zero