h2zero/NimBLE-Arduino #621: Arduino Uno R4 WiFi Compatibility

Dario-Ciceri, 2023-12-22:
Hi, I wanted to know if there is even the slightest of possibilities to use the Arduino Uno R4 WiFi BLE stack with Nimble or if I have to go directly through ArduinoBLE, I have some codes with Nimble that I would like to use but at the limit I will have to rewrite everything for ArduinoBLE.
Thanks for the wonderful work so far in any case.

h2zero, 2023-12-29:
Hello, thank you for asking. Unless drivers are available for the Uno/NimBLE then I will not be able to support it.

Dario-Ciceri, 2023-12-29:
I know that there's something called spi_dvr.h if I'm not wrong which is used to communicate with the Esp32 chip

fstrugar, 2024-01-21:
Is HCI the layer that ArduinoBLE uses to communicate with ESP32? (https://github.com/arduino-libraries/ArduinoBLE/blob/master/src/utility/ATT.cpp seems to rely on https://github.com/arduino-libraries/ArduinoBLE/blob/master/src/utility/HCI.h ?)
I don't know if that's of any help...