h2zero/NimBLE-Arduino #622: Arduino IDE ext_nimble_config.h: No such file or directory

IRGLMC, 2023-12-27:
Arduino IDE ext_nimble_config.h: No such file or directory
[NimBLE-Arduino.zip](https://github.com/h2zero/NimBLE-Arduino/files/13776810/NimBLE-Arduino.zip)
included Arduino library which does not have the file in it.

Alcalzard, 2023-12-28:
i've been experiencing the same issue.
I've also tried un/re install the lib on Arduino IDE2 or use a previous version.

h2zero, 2023-12-29:
What board are you building this for? If not building for and esp32 then you need to use the n-able platform.

cvanelteren, 2024-01-03:
Facing the same issue when building for an NRF52480

h2zero, 2024-01-09:
Please check your environment, you need to be using the n-able core for this file to be found.

OscarSheng, 2024-02-21:
> Please check your environment, you need to be using the n-able core for this file to be found.

I have same problem, I have a Arduino Uno R4 WiFi. Does this library work with this board?

h2zero, 2024-02-21:
This library will support the S3 that is on the UNO R4 WiFi, however the build environment doesn't see to have the chip defined in the config. You'll want to add these definitions to your environment somewhere:
```
ESP_PLATFORM
CONFIG_IDF_TARGET_ESP32S3
```

OscarSheng, 2024-02-23:
> This library will support the S3 that is on the UNO R4 WiFi, however the build environment doesn't see to have the chip defined in the config. You'll want to add these definitions to your environment somewhere:
> 
> ```
> ESP_PLATFORM
> CONFIG_IDF_TARGET_ESP32S3
> ```

Sorry for bothering. I'm new to this. I couldn't figure this out myself. Can you give me a more specific instruction?

h2zero, 2024-02-25:
Happy to help but I need to know more about your build environment. Are you using Arduino IDE or Platformio? If Arduino IDE I don't think there is much I can say, but in platformio you can add this:
```
build_flags = 
  '-DESP_PLATFORM'
  '-DCONFIG_IDF_TARGET_ESP32S3'
```

OscarSheng, 2024-02-26:
> Happy to help but I need to know more about your build environment. Are you using Arduino IDE or Platformio? If Arduino IDE I don't think there is much I can say, but in platformio you can add this:
> 
> ```
> 
> build_flags = 
> 
>   '-DESP_PLATFORM'
> 
>   '-DCONFIG_IDF_TARGET_ESP32S3'
> 
> ```

I'm using Arduino IDE. I thought the config should be already setup by the IDE?

h2zero, 2024-02-26:
I'm not sure how the Arduino IDE handles this board since it has 2 chips on it. I believe you would need to change the board to an esp32s3 in the IDE to compile this and upload it differently than you normally do for the main chip on this board.

richonguzman, 2024-05-17:
HI! 

I'm trying to add this to TTGO Lilygo T-Echo which is NRF52840 using platformio and I'm getting the:

`In file included from .pio/libdeps/ttgo_t_echo/NimBLE-Arduino/src/NimBLEDevice.h:18:0,
                 from src/ble_utils.cpp:1:
.pio/libdeps/ttgo_t_echo/NimBLE-Arduino/src/nimconfig.h:6:10: fatal error: ext_nimble_config.h: No such file or directory`

you mentioned using  

> Please check your environment, you need to be using the n-able core for this file to be found.

but I dont get how to add this and keep on going with this....

h2zero, 2024-05-21:
If you're using this library with an nRF device then you need to use the n-able Arduino core which will set the environment correctly.

richonguzman, 2024-05-21:
on the n-able Arduino are there examples of how to do this?

h2zero, 2024-05-22:
@richonguzman Please refer to the in instructions in the readme here: https://github.com/h2zero/n-able-Arduino

richonguzman, 2024-05-22:
> @richonguzman Please refer to the in instructions in the readme here: https://github.com/h2zero/n-able-Arduino

on the readme it appears: 

`platform = https://github.com/h2zero/platform-n-able.git#1.0.0`

should this replace the 

`platform = nordicnrf52` 

i have on platformio?

h2zero, 2024-05-22:
Yes, that is how you change the core to be able to use this library.

richonguzman, 2024-05-22:
> Yes, that is how you change the core to be able to use this library.

thanks will test it tomorrow ! ;)

Faydhax07, 2024-05-29:
Hi, I'm trying to connect an xbox controller to an arduino nano with the **_XboxSeriesXControllerESP32_asukiaaa_** library. 
And when I run the example program **_vibration_** , I get the same error: 
`Arduino\libraries\NimBLE-Arduino\src/nimconfig.h:6:10: fatal error: ext_nimble_config.h: No such file or directory`

Any thoughts?

[https://github.com/asukiaaa/arduino-XboxSeriesXControllerESP32/blob/main/examples/vibration/vibration.ino](url)

thorrak, 2024-05-29:
> Hi, I'm trying to connect an xbox controller to an arduino nano with the **_XboxSeriesXControllerESP32_asukiaaa_** library. And when I run the example program **_vibration_** , I get the same error: [...]

Which Arduino Nano do you have? The basic Arduino Nano is a different chip from the ESP32 and is not compatible with the NimBLE library, unfortunately.

Faydhax07, 2024-05-30:
I've a Nano Character3.0. Actually, I don't think it's even made by Arduino... But I bought it here : [https://fr.aliexpress.com/item/1005006075009172.html?spm=a2g0n.order_detail.order_detail_item.3.2d467d56OqErJx&gatewayAdapt=glo2fra](url)

If it is not compatible with the nimble library, can't I just install a simulator or buy a physical converter or something like that ?

thorrak, 2024-05-30:
> I've a Nano Character3.0. Actually, I don't think it's even made by Arduino... But I bought it here : [https://fr.aliexpress.com/item/1005006075009172.html?spm=a2g0n.order_detail.order_detail_item.3.2d467d56OqErJx&gatewayAdapt=glo2fra](url)
> 
> If it is not compatible with the nimble library, can't I just install an emulator or buy a physical converter or something like that ?

That is an ATMega 328-based Nano which is not compatible with this library. Those devices do not have any radios whatsoever, and thus cannot work with Bluetooth devices.

h2zero, 2024-06-05:
Closing this as I believe it to be answered.

closed 2024-06-05 by h2zero