h2zero/NimBLE-Arduino #623: how should I set an description for 2902 based code`/

JopWerff, 2023-12-29:
Hi h2zero,

I have a bit a problem getting things done with 2902 based stuff. I tried self several ways - researched a little - had a look to this keyboard stuff of sivar2311. But I didn't get it. I'm a real newbie in the area of BLE.

I searched also for examples in the NewBLE-library but nothing to find what is a bit alike what I want to do.

Can you help me?

Here my code to convert to using NimBLED from using the original BLE Library:

    NimBLECharacteristic* pCharacteristicTEMP = pBmeService->createCharacteristic(TEMP_UUID, NIMBLE_PROPERTY::NOTIFY);

    BLEDescriptor* pBLEDescriptorTEMP = pCharacteristicTEMP->createDescriptor(BLEUUID((uint16_t)0x2902));
    pBLEDescriptorTEMP->setValue("BME temperature Celsius");
    pCharacteristicTEMP->addDescriptor(pBLEDescriptorTEMP);
These last 3 lines: how I have to implement this in the NimBLE-way?

Jop

h2zero, 2023-12-29:
In this library the 2902 is automatic when the characteristic has the notification property. You need to delete those 3 lines.

JopWerff, 2023-12-30:
Not quite clear to me. How the descriptor will get its content in this "2902-automatic" way?

Here my more complete code that makes difference between Celsius and Fahrenheit, dependent of a compiler directive.

```
  // Create NimBLE Characteristics and Create a NimBLE Descriptor
  // Temperature
  #ifdef temperatureCelsius
    pCharacteristicTEMP = pBmeService->createCharacteristic(TEMP_UUID, NIMBLE_PROPERTY::NOTIFY);

    NimBLEDescriptor* pNimBLEDescriptorTEMP = pCharacteristicTEMP->createDescriptor(BLEUUID((uint16_t)0x2902));
    pNimBLEDescriptorTEMP->setValue("BME temperature Celsius");
    pCharacteristicTEMP->addDescriptor(pNimBLEDescriptorTEMP);
    
  #else
    pCharacteristicTEMP = pBmeService->createCharacteristic(TEMP_UUID, NIMBLE_PROPERTY::NOTIFY);

    NimBLEDescriptor* pNimBLEDescriptorTEMP = pCharacteristicTEMP->createDescriptor(BLEUUID((uint16_t)0x2902));   
    pNimBLEDescriptorTEMP->setValue("BME temperature Fahrenheit");
    pCharacteristicTEMP->addDescriptor(pNimBLEDescriptorTEMP);
  #endif  
```

How I have to implement this in NimBLE making a distinguish between these temperature scales?

Regards,
   Jop

h2zero, 2023-12-30:
That's is not what the 2902 descriptor is for, it's only for setting the client notification subscription status.

You'll want to use some other method to achieve what you are doing here.

closed 2024-06-13 by h2zero