h2zero/NimBLE-Arduino #627: ESP32-S3 BLE Server Vs. Detection on PC Bluetooth Dongle

wamk9, 2024-01-03:
Hi!

I need to create a project with bluetooth connection on my PC, but ANY BT library I use show only on my smartphone the connection.

My question:

Is this a caracteristic of ESP32-S3 BLE? Or I need do a extra config to work on my PC in NimBLE?

My dongle is a Baseus BT 5.0, like this:
https://pt.aliexpress.com/item/1005003601253541.html

Another thing, i try to use all examples but... Well, don't works on this situation...

h2zero, 2024-01-09:
I'm not sure what you mean, does the PC not see the advertisements from the ESP32?

wamk9, 2024-01-09:
> I'm not sure what you mean, does the PC not see the advertisements from the ESP32?

Yes, now i have success with this, but after 2 seconds or less my device disconnect on PC. See Code above:

`#include <NimBLEDevice.h>

#define SERVICE_UUID        "4fafc201-1fb5-459e-8fcc-c5c9c331914b"
#define CHARACTERISTIC_UUID "beb5483e-36e1-4688-b7f5-ea07361b26a8"

static NimBLEServer* pServer;
static NimBLEService* pService;
static NimBLECharacteristic* pCharacteristic;

/**  None of these are required as they will be handled by the library with defaults. **
 **                       Remove as you see fit for your needs                        */
class ServerCallbacks: public NimBLEServerCallbacks {
    void onConnect(NimBLEServer* pServer) {
        Serial.println("Client connected");
        Serial.println("Multi-connect support: start advertising");
        NimBLEDevice::startAdvertising();
    };
    /** Alternative onConnect() method to extract details of the connection.
     *  See: src/ble_gap.h for the details of the ble_gap_conn_desc struct.
     */
    void onConnect(NimBLEServer* pServer, ble_gap_conn_desc* desc) {
        Serial.print("Client address: ");
        Serial.println(NimBLEAddress(desc->peer_ota_addr).toString().c_str());
        /** We can use the connection handle here to ask for different connection parameters.
         *  Args: connection handle, min connection interval, max connection interval
         *  latency, supervision timeout.
         *  Units; Min/Max Intervals: 1.25 millisecond increments.
         *  Latency: number of intervals allowed to skip.
         *  Timeout: 10 millisecond increments, try for 5x interval time for best results.
         */
         NimBLEDevice::startAdvertising();
        pServer->updateConnParams(desc->conn_handle, 24, 48, 0, 60);
    };
    void onDisconnect(NimBLEServer* pServer) {
        Serial.println("Client disconnected - start advertising");
        NimBLEDevice::startAdvertising();
    };
    void onMTUChange(uint16_t MTU, ble_gap_conn_desc* desc) {
        Serial.printf("MTU updated: %u for connection ID: %u\n", MTU, desc->conn_handle);
    };

/********************* Security handled here **********************
****** Note: these are the same return values as defaults ********/
    uint32_t onPassKeyRequest(){
        Serial.println("Server Passkey Request");
        /** This should return a random 6 digit number for security
         *  or make your own static passkey as done here.
         */
        return 123456;
    };

    bool onConfirmPIN(uint32_t pass_key){
        Serial.print("The passkey YES/NO number: ");Serial.println(pass_key);
        /** Return false if passkeys don't match. */
        return true;
    };

    void onAuthenticationComplete(ble_gap_conn_desc* desc){
        /** Check that encryption was successful, if not we disconnect the client */
        // if(!desc->sec_state.encrypted) {
        //     NimBLEDevice::getServer()->disconnect(desc->conn_handle);
        //     Serial.println("Encrypt connection failed - disconnecting client");
        //     return;
        // }
        Serial.println("Starting BLE work!");
    };
[... 101 lines cut ...]
    //pAdvertising->setScanResponse(false);
    pAdvertising->start();

    Serial.println("Advertising Started");
}

void loop() {
  /** Do your thing here, this just spams notifications to all connected clients */
    //  if(pServer->getConnectedCount()) {
    // //     // NimBLEService* pSvc = pServer->getServiceByUUID("BAAD");
    // //     // if(pSvc) {
    // //     //     NimBLECharacteristic* pChr = pSvc->getCharacteristic("F00D");
    // //     //     if(pChr) {
    // //     //         pChr->notify(true);
    // //     //     }
    // //     // }
    // }
    //pServer->getConnectedCount();
    //pCharacteristic->setValue("Test loop");
    //pCharacteristic->notify();

  delay(100);
}`

I try remove notify to see if i have any diference reaction, but nothing happens...

wamk9, 2024-01-15:
Any suggestion @h2zero?

h2zero, 2024-01-17:
I would need to see the logs from the esp32 at debug level to provide much insight with this.

dimityrivanov, 2024-01-25:
@h2zero how can I enable logs ?

h2zero, 2024-06-09:
I believe this to be related to bluetooth classic, not this repo. please reopen if still having this issue.

closed 2024-06-09 by h2zero