h2zero/NimBLE-Arduino #628: NimBLEscan turn a wrong MAC adress by 2 in the last octet

kbaggen, 2024-01-03:
Hi,
I ran into an issue where when I scan by my ESP32 DevKit the MAC id is shown wrongly, e.g. it seems the BLEscan adds +2 to the last octet.

The issue seems to be reported at this link --> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/misc_system_api.html#mac-address

The code for this area:
` BLEScanResults foundDevices = pBLEScan->start(SCAN_TIME);
  deviceCount = foundDevices.getCount();
  Serial.print(deviceCount);
  Serial.println(F(" Devices Found."));
  pBLEScan->stop();
  BLEDevice::deinit(0);

  for (uint32_t i = 0; i < deviceCount; i++)
  {
    BLEAdvertisedDevice Device = foundDevices.getDevice(i);
    String DevString, DevData, DevRapt;
    DevString = Device.toString().c_str();
    DevData = DevString.substring(63);

       MAC_data = Device.getAddress().toString().c_str(); 
    Serial.println(MAC_data);`

Returns --> 78:e3:6d:3d:11:7e
Where the correct should be --> 78:e3:6d:3d:11:7**c**

h2zero, 2024-01-09:
The MAC address you are receiving is the correct BLE address. The MAC it seems you are looking to get is the esp32 base MAC, which the BLE stack will have no knowledge of.

kbaggen, 2024-01-26:
ok, I solved by just disregards the last ciffer, hence, I close it :-)

closed 2024-01-26 by kbaggen