h2zero/NimBLE-Arduino #630: How to require a passkey for Notify?

trullock, 2024-01-11:
I have a server which sends out notifications.

I want only trusted clients to be able to connect (mac unknown) so wish to protect the connection with a passkey/pin.

I can't get nrfConnect to ask for a passkey without adding READ_ENC to my characteristic.

Does adding READ_ENC imply that clients can Read?

Am I misunderstanding the API, is this a missing feature, or am I not understanding the BT standard. Must you be able to read in order to notify?

Thanks

h2zero, 2024-01-17:
To require all connections to be secure you can call `NimBLEDevice::startSecurity` in the onConnect callback.
Something like this : https://github.com/h2zero/NimBLE-Arduino/issues/92#issuecomment-720788573

trullock, 2024-01-18:
Great, thanks for your help

closed 2024-01-18 by trullock