h2zero/NimBLE-Arduino #635: Connection only on client request after disconnecting?

Pasikav, 2024-01-26:
Its more of a question than issue!

I have ESP32 project nimBLE. Basically my device is BLE Keyboard. One of the functionality is to disconnect from Server  side (esp32).
 
I made it to the point that I disconnect device using `pServer->disconnect(this->connHandle);`  and then clear ble store using `ble_store_clear()`. From the perspective of server it works fine, there is no connection but client side (windows computer) is still trying to connect, so there is no clear indication that there is no connection and also while debugging i noticed it triggers onConnect event. 

What i would like to achieve is something alike pairing mode in bluetooth headset when if you set it on pairing mode the connection is made only on client request. Is it even possible using BLE?

h2zero, 2024-01-26:
You're getting reconnected because the advertising was automatically started after disconnect. You'll want to change that setting to prevent that.

Pasikav, 2024-01-28:
But if i don't start advertising wouldn't my device be invisible for other clients or my understanding of advertising is wrong?

h2zero, 2024-06-09:
Sorry, the answer here is to advertise with a whitelist so that only bonded peers can connect. Entering "pairing mode" would be advertising without the whitelist.

closed 2024-06-09 by h2zero