h2zero/NimBLE-Arduino #636: Adafruit Feather ESP32-S3 panik resets in the Scan Continuous example.

jstraub, 2024-01-28:
Hey :) great library! It works beautifully on an Arduino ESP32 for me and I am trying to port the code to the Adafruit Feather ESP32-S3 Reverse TFT now so I can have a display. While doing so I am getting panik resets. I have narrowed this down to the following line in the Scan Continuous example:
```
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
```
The example code I am using:
```
#include <NimBLEDevice.h>

int scanTime = 5; //In seconds
BLEScan* pBLEScan;

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
  /*** Only a reference to the advertised device is passed now
    void onResult(BLEAdvertisedDevice advertisedDevice) { **/
    void onResult(BLEAdvertisedDevice* advertisedDevice) {
  /** Serial.printf("Advertised Device: %s \n", advertisedDevice.toString().c_str()); **/
      Serial.printf("Advertised Device: %s \n", advertisedDevice->toString().c_str());
    }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Scanning...");

  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value
}

void loop() {
  // put your main code here, to run repeatedly:
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
  pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory
  delay(2000);
}
```

I checked that the pBLEScan pointer is not a nullptr. If I comment out the offending line the code runs fine (but doesnt do any scanning of course). The code works fine on my other ESP32 from Arduino. 

Any idea what might be going wrong or how to debug it?

h2zero, 2024-02-06:
Could you provide a debug log and or backtrace?

h2zero, 2024-11-20:
Closing this as there has been no response, please re-open if still having this issue.

closed 2024-11-20 by h2zero