h2zero/NimBLE-Arduino #638: NimBLEServerCallbacks won't call onConnect(BLEServer*, NimBLEConnInfo&)

bmedici, 2024-02-03:
Hi,

I'm using this library in version 1.4.1 and identified an unexpected behaviour with server callbacks.

Here is the init code I'm using:
```
class ServerCallbacks: public NimBLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      deviceConnected = true;
    };

void loop() {
  log("deviceConnected: " + std::to_string(deviceConnected));
  }
}

BLEServer *pServer;
NimBLEDevice::init(BLE_NAME);
pServer = NimBLEDevice::createServer();
pServer->setCallbacks(new ServerCallbacks());
```

I implemented the server callbacks as shown and, in this situation, the `deviceConnected` variable is correctly set when my android device connects:

```
ble-master | deviceConnected: 0
ble-master | deviceConnected: 0
ble-master | deviceConnected: 0
// Android device connecting
ble-master | deviceConnected: 1
// The callback is called
ble-master | deviceConnected: 1
ble-master | deviceConnected: 1
```

On the other hand, while using the complete prototype for `onConnect(BLEServer*, NimBLEConnInfo&)`, the callback is not called at all. With:
```
class ServerCallbacks: public NimBLEServerCallbacks {
    void onConnect(BLEServer* pServer, NimBLEConnInfo& connInfo) {
      deviceConnected = true;
    };
}
```

nothing happens

```
ble-master | deviceConnected: 0
ble-master | deviceConnected: 0
ble-master | deviceConnected: 0
// Android device connecting
ble-master | deviceConnected: 0
ble-master | deviceConnected: 0
ble-master | deviceConnected: 0
// Nothing happens

```

I double checked that the only difference between both tests is changing the function prototype from
`onConnect(BLEServer* pServer)` to `onConnect(BLEServer* pServer, NimBLEConnInfo& connInfo)`.

As I need the `connInfo`parameter, this bug is a bit annoying.

Any idea ?

h2zero, 2024-02-06:
You are using v1.4.1, try switching to the master branch if you want to use the latest function signature, otherwise change to the old sig that takes the ble_conn_desc struct.

REEASD, 2024-04-29:
you can add  override.
class ServerCallbacks: public NimBLEServerCallbacks {
    void onConnect(BLEServer* pServer,BLEConnInfo& connInfo) override  {
      deviceConnected = true;
    };

void loop() {
  log("deviceConnected: " + std::to_string(deviceConnected));
  }
}

h2zero, 2024-09-24:
Closing this, please reopen if still an issue.

closed 2024-11-20 by h2zero

DanielW0830, 2025-01-10:
I had the same problem.  OnConnect (and other callbacks were not getting called)
always said : default()
turned out I had upgraded and the signature of the callback had changed.  
needed the other parameters.   
very confusing.   but it is solved.   Thank you for this thread.