h2zero/NimBLE-Arduino #640: suppressing some serial output from the scan whitelist example

linxcow, 2024-02-15:
How can i suppress following output to my serial monitor ?
20:50:32.829 -> Service Data:
20:50:32.829 -> UUID: 0xfe2c, Data:  

I just want to see the advertised lines.:
20:50:25.525 -> Advertised Device: Name:  < snipped >

          class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
            void onResult(BLEAdvertisedDevice* advertisedDevice) {
              if (!advertisedDevice->getName().empty()) {
                 Serial.printf("Advertised Device: %s \n", advertisedDevice->toString().c_str());
              }
            }
          };

h2zero, 2024-02-18:
You could just call getName and print the value. The toString method dumps more data.

linxcow, 2024-02-18:
Yes , but this was not my question.
I want to see all data on 1 line when name is not empty , some devices print on several lines.

h2zero, 2024-02-19:
You'll need to implement your own version of toString to accomplish this.

closed 2024-06-12 by h2zero