h2zero/NimBLE-Arduino #641: Support for new ESP32 Arduino core (3.x) and IDF (5.1)

shlomozippel, 2024-02-19:
I started working with the new arduino core and espidf drivers and the NimBLE-Arduino library isn't compatible anymore. There are breaking changes on the arduino side (https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html) and on the espidf side (https://docs.espressif.com/projects/esp-idf/en/stable/esp32/migration-guides/release-5.x/5.0/bluetooth-low-energy.html)

I didn't dig too deep yet but I noticed NimBLE-Arduino has it's own copy of nimble, presumably a version that has been ported to idf 4.x

h2zero, 2024-02-19:
What is it that is no longer compatible? This library doesn't really use any Arduino core functions so there shouldn't be any changes required from there. This does contain a slightly older NimBLE version but that should not matter.

shlomozippel, 2024-02-19:
When I first build it I get:
```In file included from .pio/libdeps/usba-c2/NimBLE-Arduino/src/nimble/nimble/include/nimble/nimble_npl.h:53,
                 from .pio/libdeps/usba-c2/NimBLE-Arduino/src/nimble/porting/nimble/include/os/os.h:42,
                 from .pio/libdeps/usba-c2/NimBLE-Arduino/src/nimble/nimble/include/nimble/ble.h:26,
                 from .pio/libdeps/usba-c2/NimBLE-Arduino/src/NimBLEAddress.h:23,
                 from .pio/libdeps/usba-c2/NimBLE-Arduino/src/NimBLEAddress.cpp:19:
.pio/libdeps/usba-c2/NimBLE-Arduino/src/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:457:4: error: 'esp_timer_handle_t' does not name a type; did you mean 'TimerHandle_t'?
  457 |    esp_timer_handle_t handle;
      |    ^~~~~~~~~~~~~~~~~~```

So I tried adding `CONFIG_BT_NIMBLE_USE_ESP_TIMER=n` to sdkconfig, which then resulted in a whole bunch of other errors:

```.pio/libdeps/usba-c2/NimBLE-Arduino/src/NimBLEDevice.cpp:90:62: error: 'CONFIG_BTDM_SCAN_DUPL_TYPE' was not declared in this scope; did you mean 'CONFIG_BTDM_SCAN_DUPL_TYPE_DATA'?
   90 | uint8_t                     NimBLEDevice::m_scanFilterMode = CONFIG_BTDM_SCAN_DUPL_TYPE;
      |                                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                              CONFIG_BTDM_SCAN_DUPL_TYPE_DATA
In file included from .pio/libdeps/usba-c2/NimBLE-Arduino/src/nimble/porting/nimble/include/syscfg/syscfg.h:9,
                 from .pio/libdeps/usba-c2/NimBLE-Arduino/src/nimble/nimble/include/nimble/ble.h:25,
                 from .pio/libdeps/usba-c2/NimBLE-Arduino/src/NimBLEAddress.h:23,
                 from .pio/libdeps/usba-c2/NimBLE-Arduino/src/NimBLEAdvertisedDevice.h:20,
                 from .pio/libdeps/usba-c2/NimBLE-Arduino/src/NimBLEScan.h:20,
                 from .pio/libdeps/usba-c2/NimBLE-Arduino/src/NimBLEDevice.h:22,
                 from .pio/libdeps/usba-c2/NimBLE-Arduino/src/NimBLEDevice.cpp:18:
.pio/libdeps/usba-c2/NimBLE-Arduino/src/NimBLEDevice.cpp: In static member function 'static void NimBLEDevice::init(const std::string&)':
.pio/libdeps/usba-c2/NimBLE-Arduino/src/nimble/esp_port/port/include/esp_nimble_cfg.h:13:49: error: 'MYNEWT_VAL_BLE_TRANSPORT_EVT_COUNT' was not declared in this scope; did you mean 'MYNEWT_VAL_BLE_HCI_ACL_OUT_COUNT'?
   13 | #define MYNEWT_VAL(x)                           MYNEWT_VAL_ ## x
      |                                                 ^~~~~~~~~~~
/Users/shlomo/.platformio/packages/framework-espidf/components/bt/common/api/include/api/../../../../controller/esp32c2/esp_bt_cfg.h:41:48: note: in expansion of macro 'MYNEWT_VAL'
   41 |     #define DEFAULT_BT_LE_HCI_EVT_HI_BUF_COUNT MYNEWT_VAL(BLE_TRANSPORT_EVT_COUNT)
      |                                                ^~~~~~~~~~
/Users/shlomo/.platformio/packages/framework-espidf/components/bt/include/esp32c2/include/esp_bt.h:235:33: note: in expansion of macro 'DEFAULT_BT_LE_HCI_EVT_HI_BUF_COUNT'
  235 |     .ble_hci_evt_hi_buf_count = DEFAULT_BT_LE_HCI_EVT_HI_BUF_COUNT,                     \
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/usba-c2/NimBLE-Arduino/src/NimBLEDevice.cpp:869:45: note: in expansion of macro 'BT_CONTROLLER_INIT_CONFIG_DEFAULT'
  869 |         esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/usba-c2/NimBLE-Arduino/src/nimble/esp_port/port/include/esp_nimble_cfg.h:13:49: error: 'MYNEWT_VAL_BLE_TRANSPORT_EVT_DISCARDABLE_COUNT' was not declared in this scope; did you mean 'CONFIG_BT_NIMBLE_TRANSPORT_EVT_DISCARD_COUNT'?
   13 | #define MYNEWT_VAL(x)                           MYNEWT_VAL_ ## x
      |                                                 ^~~~~~~~~~~
/Users/shlomo/.platformio/packages/framework-espidf/components/bt/common/api/include/api/../../../../controller/esp32c2/esp_bt_cfg.h:42:48: note: in expansion of macro 'MYNEWT_VAL'
   42 |     #define DEFAULT_BT_LE_HCI_EVT_LO_BUF_COUNT MYNEWT_VAL(BLE_TRANSPORT_EVT_DISCARDABLE_COUNT)
      |                                                ^~~~~~~~~~
/Users/shlomo/.platformio/packages/framework-espidf/components/bt/include/esp32c2/include/esp_bt.h:236:33: note: in expansion of macro 'DEFAULT_BT_LE_HCI_EVT_LO_BUF_COUNT'
  236 |     .ble_hci_evt_lo_buf_count = DEFAULT_BT_LE_HCI_EVT_LO_BUF_COUNT,                     \
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/usba-c2/NimBLE-Arduino/src/NimBLEDevice.cpp:869:45: note: in expansion of macro 'BT_CONTROLLER_INIT_CONFIG_DEFAULT'
  869 |         esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/usba-c2/NimBLE-Arduino/src/nimble/esp_port/port/include/esp_nimble_cfg.h:13:49: error: 'MYNEWT_VAL_BLE_MAX_PERIODIC_ADVERTISER_LIST' was not declared in this scope; did you mean 'DEFAULT_BT_LE_MAX_PERIODIC_ADVERTISER_LIST'?
   13 | #define MYNEWT_VAL(x)                           MYNEWT_VAL_ ## x
      |                                                 ^~~~~~~~~~~
/Users/shlomo/.platformio/packages/framework-espidf/components/bt/common/api/include/api/../../../../controller/esp32c2/esp_bt_cfg.h:32:56: note: in expansion of macro 'MYNEWT_VAL'
[... cut ...]
.pio/libdeps/s3/NimBLE-Arduino/src/nimble/nimble/host/src/ble_gap.c: In function 'ble_gap_log_duration':
/Users/shlomo/.platformio/packages/framework-espidf/components/log/include/esp_log.h:265:27: error: format '%d' expects argument of type 'int', but argument 6 has type 'int32_t' {aka 'long int'} [-Werror=format=]```

and more similar warnings that are treated as errors.

Feels like a combination of upgrading from idf 4.x to 5.x where some of the ble / nimble headers changed, and probably some build flags changed too

shlomozippel, 2024-02-19:
Platformio.ini to repro:

```[env]
framework = arduino, espidf
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
lib_deps = 
  h2zero/NimBLE-Arduino @ ^1.4.1

[env:usba-c2]
board = esp32-c2-devkitm-1
```

h2zero, 2024-02-20:
Oh, using this with IDF is not recommended since you have the option there to enable nimble. You will want to switch from this repo to esp-nimble-cpp.

h2zero, 2024-02-20:
> I also tested NimBLE-Arduino with arduino framework (without espidf) and still ran into the same errors with the new esp32 core v3.x, so leaving this issue open for now

This is due to using `platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5`; in that fork he has enabled NimBLE in the config, the official release will not.

h2zero, 2024-03-04:
Closing this as it is compiling and working fine with the latest Arduino core master branch. The issue here is using a third party core which provides the NimBLE core sources via IDF config.

closed 2024-03-04 by h2zero

Darmandran, 2024-03-04:
Hi, since beginning of the year i been having issues with the platformio,

```
[env:esp32dev]
platform = espressif32@6.5.0
board = esp32dev 
framework = arduino
board_build.filesystem = littlefs 
monitor_speed = 115200
```

```
08:54:35.508 > ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x40094b70
08:54:35.514 > file: ".pio/libdeps/esp32dev/NimBLE-Arduino/src/NimBLEDevice.cpp" line 879
08:54:35.519 > func: static void NimBLEDevice::init(const string&)
08:54:35.525 > expression: esp_bt_controller_init(&bt_cfg)
08:54:35.527 > 
08:54:35.527 > abort() was called at PC 0x40094b73 on core 1
08:54:35.533 > 
08:54:35.533 > 
08:54:35.533 > Backtrace: 0x40084105:0x3ffced50 0x40094b7d:0x3ffced70 0x4009a8e1:0x3ffced90 0x40094b73:0x3ffcee10 0x400f2236:0x3ffcee30 0x400d3775:0x3ffcee80 0x400d602b:0x3ffceec0 0x40105fda:0x3ffcef20
08:54:35.550 >
```
for
platform = espressif32@6.4.0 has no issues