h2zero/NimBLE-Arduino #642: [Feature request] Support for ESP32-C6

lyusupov, 2024-03-01:
ESP32-C6 support is available in Arduino Core ESP32 since version 3.x-alpha

An attempt to build the NimBLE-1.4.1 for a ESP32-C6 devkit target gives a failure:

```
In file included from /home/codespace/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32c6/include/bt/include/esp32c6/include/esp_bt.h:17,
                 from /home/codespace/Arduino/libraries/NimBLE-Arduino-1.4.1/src/NimBLEDevice.h:46,
                 from /workspaces/codespaces-blank/C6/SoftRF/software/firmware/source/SoftRF/src/platform/bluetooth/NimBLE.cpp:31:
/home/codespace/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32c6/include/bt/common/api/include/api/../../../../controller/esp32c6/esp_bt_cfg.h:19:10: fatal error: syscfg/syscfg.h: No such file or directory
   19 | #include "syscfg/syscfg.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

h2zero, 2024-03-01:
I wish this were possible but it is sadly not. Support for C6 and H2 is hindered by the way the Arduino core is configured when the bluetooth libs are built as bluedroid is enabled there and the new IDF version uses the same symbols now for the controller interface and conflicts when linked.

The only way to support these 2 chips is either by using esp-idf and Arduino as a component (recommended) or using the Arduino lib builder to disable bluedroid and nimble, controller only config.

h2zero, 2024-03-01:
Hmm, i lied... kinda. Here is a branch that compiles, no idea if it will work, probably not due to mixing of functions but give it a go and let me know.

https://github.com/h2zero/NimBLE-Arduino/tree/esp32-c6-test

lyusupov, 2024-03-02:
Thank you for development of the C6 branch !

> branch that compiles

Compilation of `NimBLE_Client.ino` demo sketch goes fine.
However, the linker is still unable to complete it's task:

```
/home/codespace/.arduino15/packages/esp32/tools/esp-rv32/2302/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld: /home/codespace/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-3662303f31/esp32c6/lib/libbt.a(os_msys_init.c.obj): in function `os_msys_init':
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/porting/nimble/src/os_msys_init.c:90: multiple definition of `os_msys_init'; /tmp/arduino/libraries/NimBLE-Arduino/nimble/porting/nimble/src/os_msys_init.c.o:/home/codespace/Arduino/libraries/NimBLE-Arduino/src/nimble/porting/nimble/src/os_msys_init.c:190: first defined here
collect2: error: ld returned 1 exit status
```

Duplicate of `os_msys_init` symbol is detected.

h2zero, 2024-03-02:
That's odd, it seemed to link fine with the latest Arduino commit in platformio.

lyusupov, 2024-03-02:
> Arduino commit in platformio.

`platformio` typically lags few versions behind Espressif official Arduino Core.
The most recent one is [3.0.0-alpha3](https://github.com/espressif/arduino-esp32/releases/tag/3.0.0-alpha3)

To reproduce, please, use official [Arduino CLI](https://github.com/arduino/arduino-cli) as follows

```
$ arduino-cli config init
$ arduino-cli config add board_manager.additional_urls https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json
$ arduino-cli core update-index
$ arduino-cli core install esp32:esp32@3.0.0-alpha3
$ arduino-cli board attach -b esp32:esp32:esp32c6
$ arduino-cli compile -v  -b esp32:esp32:esp32c6 <path-to-Nim-BLE-examples>/NimBLE_Client/
```

h2zero, 2024-03-02:
Actually I modified platformio to be able to build from the latest commit on master branch. Thanks though, I'll try that and see if I can reproduce.

h2zero, 2024-03-02:
Reproduced and fixed, please do a reset --hard on the branch and try again. I'm not expecting it to work but worth a shot. Sadly I do not have one of these devices to test on.

lyusupov, 2024-03-02:
> Reproduced and fixed

Thank you so much !

>  I'm not expecting it to work but worth a shot.

Sure!

It enters into boot loop soon after `Starting NimBLE Client` message.

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xc24
load:0x4086e610,len:0x2708
load:0x40875728,len:0x594
entry 0x4086c410
Starting NimBLE Client
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4202ef32  RA      : 0x4202d70e  SP      : 0x40816330  GP      : 0x4080c900
TP      : 0x40808e60  T0      : 0x7fffffff  T1      : 0x7f7f7f7f  T2      : 0xffffffff
S0/FP   : 0x4080f7e0  S1      : 0x00000000  A0      : 0x4080f7e0  A1      : 0x00000000
A2      : 0x0000000c  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000
A6      : 0x4080d848  A7      : 0x00000000  S2      : 0x4080f000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x4087dc9c  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000005  MTVAL   : 0x000000a0
MHARTID : 0x00000000

Stack memory:
40816330: 0x00000000 0x4080f50c 0x4080f000 0x4202d70e 0x00000000 0x4080f000 0x4080d000 0x420118e4
40816350: 0x00000000 0x4080f000 0x4204a800 0x420105c2 0x00000000 0x4080e218 0x40816394 0x42018898
40816370: 0x00000000 0x00000000 0x40816394 0x42003c14 0x00000000 0x4080f000 0x4080e000 0x420009b6
40816390: 0x00000000 0x4081639c 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x7b327396
408163b0: 0x00000000 0x00000000 0x00000000 0x4201a5a6 0x00000000 0x00000000 0x00000000 0x00000000
408163d0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
408163f0: 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x0006620d 0x4080f834 0x4080f834 0xf9d492cb 0xfcff26da
40816410: 0xdc7243f8 0xdf71a967 0xfc44fc78 0x9999ac8e 0x06808841 0xfe248a2d 0x0b4f5825 0xf4063aef
40816430: 0x1d9752f1 0x869cf2d4 0x5da1be49 0x9f755986 0x51083001 0x2379953f 0x81328892 0xd1ab8074
40816450: 0x03f4a0a7 0x690444b8 0x24356ff6 0x226858ce 0x32260632 0xb11c67ca 0x680a3022 0xdf2cca2f
40816470: 0xf95e4fde 0xd6e0d883 0xb74cbd69 0x23966df6 0xec862932 0x5841f7e8 0x32e167a1 0x30f9aaf4
40816490: 0x78a97ea6 0x43091f53 0xfbb936ff 0xaca4fd3c 0x45b03ca9 0x8220f123 0x138a0398 0x38553bba
408164b0: 0xa3ee933e 0xa90131c8 0xb3e08e8e 0xc16d2346 0xfa890210 0xbbeb0a5f 0x97892ee0 0xec923ff4
408164d0: 0x677c864e 0x98242ee4 0x5562ae05 0x8b84c080 0x86fc0784 0x738e63b2 0x486a2029 0xd7fdd17f
408164f0: 0x8d7a9fe1 0x405e8441 0x7cccffd3 0x8f10d62d 0x4762a425 0x791379c0 0x287be90b 0x1bbf6761
40816510: 0x934a9748 0x40a754c5 0xf452ffd8 0x8accb9a2 0xa911a824 0x70366b91 0x39025b44 0xd3b78f9d
40816530: 0x8ac37f23 0x8c3fc266 0xdefab16e 0xf6aac362 0x3ee0a60b 0x77d18906 0x1e490c6b 0xacb6f75f
40816550: 0x70a66346 0xd5aa637a 0x857933be 0xc1cb51a9 0xd7a6191c 0x74ada178 0x5c8306e7 0xdd2c32ff
40816570: 0xf99de104 0xd7e72906 0xdf6229a7 0x241006e5 0xd6a5468c 0xb015f319 0xb79a4003 0x54202ce7
40816590: 0xfdfac8ea 0xbc817082 0xff7b61d4 0xa340f514 0x2d4c65ef 0xfd2f5468 0x1321469e 0x0cb9dc57
408165b0: 0x3779bcfd 0x84423da8 0x689c3ba8 0x2b731314 0xcbab189a 0x2b33fa2a 0x3542521c 0x6d1958e1
408165d0: 0x05d6ef56 0x61a4f219 0xff626504 0x1fdf4736 0xe91b4afd 0xa32fcf07 0x9b0ada1c 0x0e73f34e
408165f0: 0xf40a7dff 0xefae4843 0x62be59ab 0x1c009467 0x21e0018e 0xf197b82d 0xf2000451 0xa4567d66
40816610: 0x7dbe9d74 0x56ca0875 0xf65d77c3 0x21742483 0x0a48a132 0xed819bb8 0xc3367098 0xcd1aade8
40816630: 0x2fb6f999 0x1660a2da 0x1d26ff7f 0x7b20d040 0x06128a00 0x0d712a97 0x1e6844cb 0xf492ae8a
40816650: 0x2374e994 0x87900a0a 0xd0f63c48 0x0a622010 0x18d49a2d 0x5e8239e0 0x0b9a2989 0x45a94caf
40816670: 0x09164e9e 0x64528bb5 0x97c7f3cb 0x08965fe1 0x64742c1d 0x47bbbe5c 0x258354c6 0x353f622f
40816690: 0x1b66e46f 0x0060541c 0x5bf2f8c6 0xa122cf91 0x902723cc 0x256b5e14 0x57c120e6 0x09afe630
408166b0: 0x404f23fb 0xf0a80830 0x398ffa11 0xfe70aa27 0x8117a230 0xdbbac27b 0xfb202c0c 0xb865be65
408166d0: 0xcf0307db 0x54b08e47 0xcbbbbbda 0x08d49547 0x30c44821 0xa8722b6f 0x3a4dc0d1 0xea3626d1
408166f0: 0x4d372a45 0xbd431851 0x04046460 0x78902593 0xd6a11493 0xc59ef5a3 0x17333308 0x2e77ff46
40816710: 0xf6dfbd66 0x6e1c57a8 0xe5527128 0x1c748635 0x3a4b0717 0xdf85fd78 0xd488e1d6 0xd87d8a34

ELF file SHA256: 4ea8ee244b00f9a7

Rebooting...
```

Will try to decode the backtrace later....

h2zero, 2024-03-02:
lol, not surprised at all. Mixing code is what's happening here. The real fix for this will be to rename A LOT of functions and hope that's enough.

h2zero, 2024-03-04:
I've just pushed an update that renames all the conflicting symbols. Please do a `reset --hard` on the branch and let me know what happens.

lyusupov, 2024-03-04:
> let me know what happens

A little bit different PC and SP, but the fault is of the same type

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0xc (SW_CPU),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4001975a
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xc24
load:0x4086e610,len:0x2708
load:0x40875728,len:0x594
entry 0x4086c410
Starting NimBLE Client
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4202ee12  RA      : 0x42010694  SP      : 0x40815c40  GP      : 0x4080c240
TP      : 0x408086b0  T0      : 0x40022494  T1      : 0xffffffe0  T2      : 0xffffffff
S0/FP   : 0x4080d000  S1      : 0x4080f000  A0      : 0x42011682  A1      : 0x00000000
A2      : 0x42010050  A3      : 0x00000000  A4      : 0x4080f000  A5      : 0x00000000
A6      : 0x4080cf90  A7      : 0x00000000  S2      : 0x00000000  S3      : 0x00000000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x4087dc9c  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40800001  MCAUSE  : 0x00000007  MTVAL   : 0x0000002c
MHARTID : 0x00000000

Stack memory:
40815c40: 0x00000000 0x4080db58 0x40815c74 0x42018806 0x00000000 0x00000000 0x40815c74 0x42003c00
40815c60: 0x00000000 0x4080f000 0x4080e000 0x420009b6 0x00000000 0x40815c7c 0x00000000 0x00000000
40815c80: 0x00000000 0x00000000 0x00000000 0x72aefd79 0x00000000 0x00000000 0x00000000 0x4201b720
40815ca0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
40815cc0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xbaad5678 0x0006692d
40815ce0: 0x4080f114 0x4080f114 0xe3f9edfb 0xed228268 0x7bcdbac7 0x24bc3af3 0x48620e5c 0xdbfce97c
40815d00: 0xc9ffafaa 0xb546c815 0x8d82a09f 0xcfa0db87 0x5b140411 0x57ede3a6 0x24a2a105 0x97e2a4ad
40815d20: 0x47fae67f 0x7ef629b7 0x59effff1 0xe5a37254 0x496dde2a 0x666ddf48 0x32c8ed4c 0x17c145c8
40815d40: 0x08fcbf7f 0x0556c923 0x1bfe2b63 0xd148b991 0x01ab174e 0x74f98de7 0x6499d04d 0x36daa4fd
40815d60: 0x236af6aa 0xf408daa7 0xeee5b9b9 0x56e7f96d 0x7006b127 0xbdd39cc5 0xebc913a7 0x6f5bcba1
40815d80: 0x8a439c73 0x606cc05a 0xe464de9f 0x565256c4 0x720a69ba 0xed2fa779 0x8d347611 0xaa779171
40815da0: 0xe46353fa 0xe917ad3e 0xbb31376b 0x1a31e9dc 0xec882768 0xe15abe5f 0xfd2aecea 0x5cd1ccda
40815dc0: 0xc3681728 0xcae558eb 0x7c8db2eb 0x1310bc4d 0x6483c5b1 0xf5249069 0x76299000 0x52e5bf16
40815de0: 0x8827b579 0x34b96824 0xdef78ace 0x4b68cbc9 0x3e83e1e6 0xf5655e33 0x08334248 0x8fda2fa2
40815e00: 0xffa5a83d 0x6da00395 0xe632c4b8 0xc16333dc 0x1d3695b3 0xc7b2aabc 0x029da173 0x20c75bd7
40815e20: 0x7cfa836d 0x05802351 0xb72ec5da 0x272a00cd 0xbc08c336 0x483e6e24 0x2c34c6d0 0xee46a4cd
40815e40: 0xe767ffe6 0xc65798b5 0xc1afbcd2 0x1b0fea35 0x48c38078 0xfea8d9b3 0x7a1d44ab 0x2b55609d
40815e60: 0xcb51adfb 0x018d904e 0x5de3f9de 0x8026341c 0xa3c0bc2a 0x23da77f6 0x9d410db8 0x59fffea8
40815e80: 0xde5d4df2 0xd9948007 0xef3dfdea 0xc55dd0ca 0xfc269aa2 0xf9763f1d 0x8cfc98aa 0xeca72431
40815ea0: 0xcc4f9f21 0xdc0d2352 0x5cdf95a7 0xb481e182 0x5a0a3898 0xd2a9ade1 0x80124897 0x510fc67d
40815ec0: 0xf5ea9df6 0x2871434b 0x13db84fa 0x6dc1ba6c 0x20d88b61 0xcc59159f 0x51f3820d 0xb53fe7e0
40815ee0: 0x8fa918ad 0x969570ff 0xcd5dbfef 0x83787ee0 0x55262240 0xa5d61cfb 0x410648b8 0xb46b1fc2
40815f00: 0x55531faf 0xb00298bc 0x5ad59665 0x0bf048ac 0x12782332 0xdd43bf68 0x96230d8b 0x92bcfe63
40815f20: 0x37a8f91c 0xe49d6cd4 0xfff4e98a 0xeb2adeb8 0xabc14b02 0xe5fb5a98 0x86b17210 0x5fa9dd7a
40815f40: 0x99f4fde6 0xf40230d7 0x3e09dde1 0x35ef0a68 0x54849891 0x3de55910 0x180b1049 0x8db99db0
40815f60: 0x735c5b3c 0x210476dc 0x9ecdfcc5 0x09ec5ec7 0xac518a91 0xad24717f 0xf5e81051 0x8f95c3fd
40815f80: 0x9ffd4ed9 0x47b09342 0xebabffab 0x550eebad 0x21199222 0x77befc8b 0x5b4700e1 0x5cf1bb7b
40815fa0: 0x57ffb7cd 0x738a9eb4 0x3f6ca2d1 0x843326c8 0x7416296d 0x679d5937 0xcb034b61 0xd7e6f13f
40815fc0: 0x5d7e19dc 0x12f44c07 0x51afb3cd 0x09e97a95 0x135003c6 0x2ecbb0f6 0xd50b6602 0xe2edb777
40815fe0: 0xddeee05c 0x53337da1 0x3a460ac7 0x87e53055 0xfd2c08ba 0x5dc7c36d 0x6a08602a 0xba99f59a
40816000: 0x12c4212a 0xd5dd56c5 0xeef22830 0x1c34efc9 0xef053834 0xa123c110 0x9e3f9a29 0xc550d384
40816020: 0x2e0150fd 0x6837adae 0xc383c211 0xf0bf8ee3 0x1bc8a927 0x2440247e 0xbd8eda3d 0x886a65c6

ELF file SHA256: 03966be418b8f485

Rebooting...

```

h2zero, 2024-03-04:
Thanks,that's too bad. Definitely need a backtrace to progress. I'll try to get my hands on a board to test with soon.

lyusupov, 2024-03-05:
My ESP32-C6 board does not have USB/JTAG (GPIO12/13) pins exposed, so capturing a backtrace with assistance of OpenOCD will be problematic for a while...

I found out that ESP32-**C3** build of the same sketch _( NimBLE_Client.ino )_ with Arduino Core **3.0.0-alpha3** does also produce a binary that fails to boot properly.

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x38c
load:0x403cc710,len:0x6b0
load:0x403ce710,len:0x25b0
entry 0x403cc710
Starting NimBLE Client
E (206) BLE_INIT: hci inits failed

E (206) BLE_INIT: nimble host init failed

assert failed: xQueueReceive queue.c:1475 (( pxQueue ))
Core  0 register dump:
MEPC    : 0x40381b32  RA      : 0x4038563c  SP      : 0x3fc98b50  GP      : 0x3fc8d200
TP      : 0x3fc89d20  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130
S0/FP   : 0x00000001  S1      : 0x3fc98cb7  A0      : 0x3fc98bb4  A1      : 0x3fc8d41d
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc90000
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000090  S3      : 0x3fc98ba8
S4      : 0x3fc98ba8  S5      : 0x4038606a  S6      : 0x00000000  S7      : 0x3c057bd2
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938
MSTATUS : 0x00001801  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fc98b50: 0xa5a5a5a5 0xa5a5a5a5 0x3c05800c 0x4038b5be 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x3fc8d41c
3fc98b70: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x35373431 0xa5a5a500 0x3fc8d754 0x3c058170 0x3fc8d738
3fc98b90: 0x3c057bd2 0x3fc8d764 0x3fc98b7c 0x3fc8d768 0x3c05800c 0x3fc8d41c 0x00000000 0x00000000
3fc98bb0: 0xa5000000 0x65737361 0x66207472 0x656c6961 0x78203a64 0x75657551 0x63655265 0x65766965
3fc98bd0: 0x65757120 0x632e6575 0x3734313a 0x28282035 0x51787020 0x65756575 0x00292920 0xa5a5a5a5
3fc98bf0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc98c10: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc98c30: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc98c50: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc98c70: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x3ec43b77 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000000
3fc98c90: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc90000 0x3fc8ed04 0x3fc90024 0x4038606a
3fc98cb0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xffffffff 0x42003834 0x00000000 0x00000000 0x3fc8d200
3fc98cd0: 0x3fc89d20 0x00000000 0x00000000 0x00000000 0x3fc90000 0x3fc8ed04 0x3fc90024 0x42019daa
3fc98cf0: 0x00000000 0x00000000 0x00000000 0xffffffff 0x00000000 0x00000000 0x00000000 0x00000000
3fc98d10: 0x00000000 0x00000000 0x00000000 0x40380298 0x00000000 0x00000000 0x00000000 0x42003840
3fc98d30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98d50: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc98d70: 0xbaad5678 0x00000160 0xabba1234 0x00000154 0x3fc98cc0 0x00000000 0x3fc8f3ec 0x3fc8f3ec
3fc98d90: 0x3fc98d80 0x3fc8f3e4 0x00000004 0x00000000 0x00000000 0x3fc98d80 0x00000000 0x00000015
3fc98db0: 0x3fc97d70 0x626d696e 0x685f656c 0x0074736f 0x00000000 0x00000000 0x3fc98d60 0x00000015
3fc98dd0: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc909d0 0x3fc90a38 0x3fc90aa0 0x00000000
3fc98df0: 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x420373c6 0x00000000 0x00000000
3fc98e10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98e30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98e50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98e70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98e90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98eb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98ed0: 0x00000000 0xbaad5678 0x00027121 0x3fc902a4 0x3fc902a4 0x00000000 0x00000000 0x00000000
3fc98ef0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98f10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc98f30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: fd632cc1ac790887

Rebooting...
```

<br>
<br>

So if you have ESP32-C3 readily available - you can try it first `( with 3.0.0-alpha3 )`.

h2zero, 2024-03-05:
Interesting, thanks! Looks like the queue is not initialized for these boards... I'll look into that

h2zero, 2024-03-07:
@lyusupov,I just pushed a few commits to this, seems to work for me  😄

lyusupov, 2024-03-07:
@h2zero 

Thank you!

Yes, it looks good to me as well, both with ESP32-C3 and ESP32-**C6**:

```
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x4086c410,len:0xc24
load:0x4086e610,len:0x2708
load:0x40875728,len:0x594
entry 0x4086c410
Starting NimBLE Client
controller lib commit: [77d09ce]
Advertised Device found: Name: , Address: xx:xx:xx:xx:xx:xx, manufacturer data: 0600010920029cbc0cecd8caad59d727965b13de6546c8da802af2038f
Advertised Device found: Name: HONOR Band 5i-406, Address: xx:xx:xx:xx:xx:xx, manufacturer data: 7d02010300ffff, txPower: 4
                                                                                                                           Service Data:
                                                                                                                                        UUID: 0x3802, Data: 8▒▒▒
Advertised Device found: Name: Galaxy Watch (CC1E) LE, Address: xx:xx:xx:xx:xx:xx, appearance: 192, manufacturer data: 750001000200010102
```

I suppose that your recent development efforts are sufficient to consider this GitHub ticket as 'complete'.
I am ready to close this issue report.

If me or someone else will face up with another ESP32-C6 NimBLE issue - we should open a new ticket.

h2zero, 2024-03-07:
Sounds good, thanks! 

Going to need to test this for some time before merging into master.

NeilvdM, 2024-07-30:
Hello @h2zero , please accept my apology in advance for asking this in this thread, but I need some guidance/direction.

I have a DFRobot ESP32-C6 board, which I would like to use to send sensor data over BLE to a mobile application.

Let me give you a short background.. I do my hardest to learn and figure out things myself, (where possible), but Im certainly still a novice when it comes to coding. 
I started in Arduino IDE with the IBeacon example sketch, got that working. My next step was to figure out how to send the sensor reading, instead of the "hello" over the broadcast ([that how I ended up on this thread](https://github.com/h2zero/NimBLE-Arduino/issues/215)), and discovered NimBLE. 
When running the NimBLE Client example, or any of the others, I get the "syscfg/syscfg.h" error, and that is how I ended up on this thread.
And this is where I feel I have gone down a very dark rabbit hole where I have installed VCCode and PlatformIO that you guys have been talking about in this thread and even now attempting the Platformio build that you and Jason from Tasmoto has been working on ..

As you can see, I might have complete gone off in the wrong direction.

If this is something that you could help me with, I would be grateful and if there is another way we could communicate, other than me spamming this thread, please let me know.

Thank you!!

h2zero, 2024-07-30:
Hello @NeilvdM, good for you for trying to sort this out and learn. You've gone in the right direction, it's just a little steeper of a learning curve 😄 

The C6 is not able to be supported with this repo, but instead [esp-nimble-cpp](https://github.com/h2zero/esp-nimble-cpp) needs to be used, which means you will need to use esp-idf and Arduino as a component of that to get the Arduino interfaces. Or alternatively we can use a for of the Arduino core that @Jason2866 has created for use in platformio.

I would suggest starting with the core that tasmota uses: https://github.com/Jason2866/platform-espressif32 follow the readme to setup the environment. 

Let me know how you get on with that and I'll try to help further.

h2zero, 2024-08-01:
You're getting close 🙂! 

As for your question, yes you could use that ibeacon example and work from there, that will use the default BLE library in the official arduino core. This library is designed to be a lighter weight and more capable/stable replacement for that. What you should do depends on your needs.

NeilvdM, 2024-08-01:
obviously, the lighter weight and stability will be important for me .. so then what you are telling me is that I need to learn VSCode.

Then that brings me to these points:
```
[env:platformio]
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
framework = arduino
board = esp32-c6-devkitc-1
```
is this correct?
Do I need to add your library now below the board?
`LIB_DEPS = h2zero/NimBLE-Arduino@^1.4.0`

Then start playing with the example sketches?
Why when I copy an Example sketch from the NimBLE Github, for example, I get parsing errors in VSCode?

Would you be willing to share an email address that I can contact you on? (thats if you dont mind)

THANK YOU FOR THE INFO YOU HAVE SHARED SO FAR

h2zero, 2024-08-01:
You'll certainly need to learn to use VScode and platformio for this, you'll be glad you did later 😄 

This is what you need:
```
[env:platformio]
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
framework = arduino
board = esp32-c6-devkitc-1
lib_deps = https://github.com/h2zero/esp-nimble-cpp@^1.4.0
```

Then you can use the examples from here, you might have to add `#include <Arduino.h>` at the top of the file to get it to compile.

I prefer to keep the communication here so others can benefit.

NeilvdM, 2024-08-03:
Well I found the reason for my Parsing errors .. I was inserting the sketch in the .ini file, not the main.c :) .. lol
Now on to the next step .. I include the `#include <Arduino.h>` at the top line of main.c file, but I get the error 
"message": "cannot open source file \"Arduino.h\"",

I followed an Indestructible example of getting an Arduino sketch to run in Platformio, the start of that .ini looks like:

```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino

monitor_speed = 115200
board_build.partitions = huge_app.csv
lib_deps =
  https://github.com/trevorwslee/Arduino-DumbDisplay
  tanakamasayuki/TensorFlowLite_ESP32@^1.0.0
build_flags =
  -DFOR_ESP32
```

That main.c also have #include <Arduino.h> as the first line, but that does not give the error

What am I missing?

NeilvdM, 2024-08-04:
ok, so to see if I can get the basics working, I configured my ini file as such:
```
[env:platformio]
platform = https://github.com/Jason2866/platform-espressif32.git#Arduino/IDF5
framework = arduino
board = esp32-c6-devkitc-1
lib_deps = https://github.com/h2zero/esp-nimble-cpp@^1.4.0
```

Then in the main.ccp I followed your beginner example to create a BLE server

```
#include NimBLEDevice.h

// void setup() in Arduino
void app_main(void)
{
    NimBLEDevice::init("NimBLE");
    
    NimBLEServer *pServer = NimBLEDevice::createServer();
    NimBLEService *pService = pServer->createService("ABCD");
    NimBLECharacteristic *pCharacteristic = pService->createCharacteristic("1234");
    
    pService->start();
    pCharacteristic->setValue("Hello BLE");
    
    NimBLEAdvertising *pAdvertising = NimBLEDevice::getAdvertising();
    pAdvertising->addServiceUUID("ABCD"); 
    pAdvertising->start(); 
}
```

But this is the error that Im getting now:

> Processing * in esp32-c6-devkitc-1 environment
> --------------------------------------------------------------------------------------------------------------------------------------------
> Building & Uploading...
> Library Manager: Installing throwtheswitch/Unity @ ^2.5.2
> Unpacking  [####################################]  100%
> Library Manager: Unity@2.6.0 has been installed!
> Error: Nothing to build. Please put your test suites to the 'D:\OneDrive\Personal\Documents\PlatformIO\Projects\ESP32C-6\test' folder
> Uploading stage has failed, see errors above. Use `pio test -vvv` option to enable verbose output.
> --------------------------------------------------------------------- esp32-c6-devkitc-1:* [ERRORED] Took 13.14 seconds ------------
> 
> ======================= SUMMARY 
> Environment         Test    Status    Duration
> ------------------  ------  --------  ------------
> esp32-c6-devkitc-1  *       ERRORED   00:00:13.145

At this rate, it feels like Im never gonna get any where

mhaberler, 2024-08-04:
@NeilvdM I suggest you consider the pioarduino fork which I have nicely working with NImble-Arduino master [here](https://github.com/mhaberler/sensorbox) 

start here: https://github.com/espressif/arduino-esp32/discussions/10039 - maybe check the discord channel
to here: https://github.com/pioarduino/platform-espressif32

NeilvdM, 2024-08-05:
> @NeilvdM I suggest you consider the pioarduino fork which I have nicely working with NImble-Arduino master [here](https://github.com/mhaberler/sensorbox)

you are joking with this right? :) I just opened that page and its like Senior Varsity stuff, that is like throwing me at the bottom of the Mariana Trench and say, "alright, there you go, swim to the top" .. lol
There is so much information there, I dont know where to look at.

But In any case, Ill check out the other links you suggested.

Thanks

h2zero, 2024-08-10:
@NeilvdM Sorry for the delay, it looks like you have some issue with the folder arrangement, I will email you.

Jason2866, 2024-08-10:
I made an example. It is for the esp32 for the esp32-c6 it should work too. With only replacing the board against a c6 one.

https://github.com/pioarduino/platform-espressif32/tree/develop/examples/espidf-arduino-h2zero-BLE_scan

lyusupov, 2024-08-10:
I've re-opened this ticket due to high message traffic.
@h2zero please, feel yourself free to close it again when it is done.

troglodisme, 2024-08-20:
I would also love to have a working solution for this. +1

dncex, 2024-09-23:
i am sorry but I am a little bit confused. Is it a chance now to use nimBLE with arduino framework and esp-c6? (not arduino as an idf-component). I've tried to compile the release 1.4.2 , using: platform = https://github.com/pioarduino/platform-espressif32/releases/download/51.03.04/platform-espressif32.zip
With these configuration , it doesn't compile, but @mhaberler mentioned he succeded in.

Jason2866, 2024-09-23:
Currently working on making it easier to compile Arduino projects with IDF using Platformio.
With a slightly modified IDF (just warning settings changed) project Tasmota compiled without changing a single line of code with Arduino/espidf.

EmileSpecialProducts, 2024-10-13:
> It is not currently possible to use the C6 with this library and Arduino core.

The same is for the ESP32-H2, and the Android 3.1.0RC1 :pleading_face:

avaksru, 2024-12-13:
Where did the branch https://github.com/h2zero/NimBLE-Arduino#esp32-c6-test go?

h2zero, 2024-12-13:
Hello @avaksru that branch was just for testing and has been deleted. I've aggregated the changes from the test branches into 1 here: https://github.com/h2zero/NimBLE-Arduino/tree/c6-build

Not sure what you mean by everything worked? The only thing that is working is scanning.

avaksru, 2024-12-13:
Thanks for the quick response. The scan was working. Everything was assembled from the #esp32-c6- test branch without errors. And it worked great! But from the #c6-build branch, the build goes through with errors.

`.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld.exe: .pio/build/esp32c6_4mb/libb1c/libNimBLE-Arduino.a(ble_hs.c.o): in function `ble_hs_tx_data':

.pio/libdeps/esp32c6_4mb/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:584: undefined reference to `ble_hci_trans_funcs_ptr'

.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld.exe: .pio/libdeps/esp32c6_4mb/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs.c:731: undefined reference to `ble_hci_trans_funcs_ptr'

.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/12.2.0/../../../../riscv32-esp-elf/bin/ld.exe: .pio/build/esp32c6_4mb/libb1c/libNimBLE-Arduino.a(ble_hs_flow.c.o): in function `ble_hs_flow_startup':
.pio/libdeps/esp32c6_4mb/NimBLE-Arduino/src/nimble/nimble/host/src/ble_hs_flow.c:260: undefined reference to `ble_hci_trans_funcs_ptr'
....`

h2zero, 2024-12-13:
It builds fine with:
```
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#3.0.1
  framework-arduinoespressif32-libs @ https://github.com/espressif/arduino-esp32/releases/download/3.0.1/esp32-arduino-libs-3.0.1.zip
```

Keep in mind that this is just a test area and that this branch is not supported for use.

Jason2866, 2024-12-13:
@avaksru Fully working is using h2zero IDF NimBLE and compile it with Arduino as a component of IDF. See this example https://github.com/pioarduino/platform-espressif32/tree/Arduino/IDF53/examples/espidf-arduino-h2zero-BLE_scan
To use change in the example platformio.ini the platform entry to:
`platform = https://github.com/pioarduino/platform-espressif32.git#Arduino/IDF53`
Mandantory the `idf_component.yml`!

pierreyves258, 2024-12-29:
Same as @avaksru with arduino-esp32 v3.1.0
Reverting to 3.0.1 works fine tho

@Jason2866 the link https://github.com/pioarduino/platform-espressif32/tree/Arduino/IDF53/examples/espidf-arduino-h2zero-BLE_scan is dead

Jason2866, 2024-12-29:
@pierreyves258  Yes, it is merged in branch master https://github.com/pioarduino/platform-espressif32/tree/main/examples/espidf-arduino-h2zero-BLE_scan

h2zero, 2025-02-05:
Nice to see they're making changes to benefit more people.

I was starting to work on re-writing the driver to make this repo work with the c6, I guess that won't be needed now.

Any interest in pico w support?

DennisMart, 2025-02-17:
Hello, I would like to implement NimBLE-Arduino in my project, but I am using the ESP32-C6 and would like to know if it is already compatible with this chip or what alternatives I have.

Thank you in advance.

Jason2866, 2025-02-17:
@DennisMart It is in the works https://github.com/espressif/arduino-esp32/pull/10937

Alternatively you can build Arduino as an component of IDF and use the h2zero IDF lib.
See example https://github.com/pioarduino/platform-espressif32/tree/main/examples/espidf-arduino-h2zero-BLE_scan

lyusupov, 2025-02-20:
https://github.com/espressif/arduino-esp32/discussions/10991

h2zero, 2025-04-22:
@mhaberler @pierreyves258  @lyusupov @troglodisme @NeilvdM 
#939 is open and working, please check it out and let me know if anything doesn't work.

lyusupov, 2025-04-23:
@h2zero 

build 
- for C6 
-  with Core 3.2.0 
-  and NimBLE taken from this branch ( https://github.com/h2zero/NimBLE-Arduino/tree/nimble-1.6-c6-build-working ) 

gives a failure:

```
/home/codespace/.arduino15/packages/esp32/tools/esp-rv32/2411/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: /home/codespace/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.4-2f7dcd86-v1/esp32c6/lib/libbt.a(hci_driver_standard.c.obj):(.data.hci_driver_vhci_ops+0x0): multiple definition of `hci_driver_vhci_ops'; /tmp/arduino/libraries/NimBLE-Arduino-nimble-1.6-c6-build-working/nimble/esp_port/esp-hci/src/objs.a(na_hci_driver.c.o):/home/codespace/Arduino/libraries/NimBLE-Arduino-nimble-1.6-c6-build-working/src/nimble/esp_port/esp-hci/src/na_hci_driver.c:62: first defined here
collect2: error: ld returned 1 exit status
```

h2zero, 2025-04-23:
Weird, it builds perfectly in the CI, wonder what could be different?

lyusupov, 2025-04-23:
No.
However, when I comment this code, build goes fine:

[image]

h2zero, 2025-04-23:
> No. However, when I comment this code, build goes fine:
> 
> [image]

That's exactly what I was going to do 😄 Might was well use the instance from the Arduino binary.

h2zero, 2025-04-23:
@Jason2866 I am also using pioarduino in the CI builds without issue, not sure what is happening in this case, but the fix is easy enough.

lyusupov, 2025-04-23:
Ok. Looks like it was my local issue.
I had both 

```
< ... skipped .. >

#include <BLEDevice.h>

< ... skipped .. >

#include <NimBLEDevice.h>

< ... skipped .. >
```
in  my sketch.

When I got rid of `#include <BLEDevice.h>` - the build has completed successfully.

h2zero, 2025-04-23:
Cool, thanks! I actually decided to rename that struct to ensure the correct functions are used and prevent this sort of thing in the future.

mhaberler, 2025-04-23:
tried s3, c3, c6 BLE5-extended ad examples - no issues so far

btw debugging c6 now works great with this [openocd](https://discord.com/channels/1263397951829708871/1263397951829708874/1364628921638719488)  and pioarduino 3.2

lyusupov, 2025-04-23:
@h2zero 

The branch has 1.6 index while mainline NimBLE is 2.x.x

[image]

<br>

**what does 1.6 index mean ?**

h2zero, 2025-04-23:
It's a reference to the underlying NimBLE version: https://github.com/espressif/esp-nimble

lyusupov, 2025-04-23:
@h2zero 
Ryan, I have a minor question about resource usage.

An Arduino (Core 3.2.0) sketch that I build with NimBLE and ArduinoBLE  for ESP32-C6 gives
- approx. 1.6 Mbytes binary with ArduinoBLE
- approx. 2.0 Mbytes binary with NimBLE 

How much of debug stuff is still enabled  in the `nimble-1.6-c6-build-working` branch ?
Are there any chances to get less binary in size when you will merge this branch into upcoming Release ?

h2zero, 2025-04-23:
Unfortunately that is the punishment we take for using the stl here more than anything to do with NimBLE. In fact I'd bet that if you just used the C lib without the C++ wrapper it would be smaller than any other library.

Some things to do are make sure all logs are disabled and make sure you are disabling any BLE roles that you don't use, sever or client etc.

lyusupov, 2025-04-24:
I've re-built and checked the ESP32-C6 binary sizes again today and got
* 1543 Kbytes binary with ArduinoBLE
* 1574 Kbytes binary with NimBLE
* 1980 Kbytes binary with  Bluedroid

Quite good result for NimBLE, I guess.

Since the ESP32-C6 feature is no longer an issue - I am closing this ticket now.
Feel yourself free to re-open or clone when necessary.

closed 2025-04-24 by lyusupov

h2zero, 2025-04-24:
Glad it's only about 30k difference, if you use mbedtls for anything in your code (AWS, HTTPS?) then you can get it a little smaller by enabling the mbedtls override for tinycrypt.

Jason2866, 2025-04-24:
@h2zero mbedtls is always used in Arduino Core when wifi is used. The needed ciphers for WPA2 / WPA3 handshake are "powered" by mbedtls. So using mbedtls should be a default when wifi is used.

[1 more reply]

PR #939 (2025-04-24): Add support for  esp32c6, esp32h2, esp32c2, esp32c5 chips.
Closes #642  and #944